kriswiner/MPU9250 #236: euler angle

HwangHanGyeol, 2018-02-20:
Hello.
I need your help.
Can I convert the values from the MPU9250 directly into euler angles?
Or convert to quaternions?
I want to do without MPU9250 dmp.

kriswiner, 2018-02-20:
Try this:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

HwangHanGyeol, 2018-02-20:
Thank you for your reply.
There is a question.
When you enter the link that you attached, there is 'MadgwickQuaternionUpdate' in the function.
Is this part going to get a quaternion with the values from each sensor?

kriswiner, 2018-02-20:
Yes, add it to the tab as an adjunct to the main sketch.

HwangHanGyeol, 2018-02-20:
I saw the file.
But, the board is different I was running the 'MPU9250BasicAHRS.ino' file you wrote.
Fortunately, the code works well, but the yaw value is in the range of 10 to 60.

HwangHanGyeol, 2018-02-20:
For reference, I use the following quaternion function.
MahonyQuaternionUpdate (ax, month, az, gx * PI / 180.0f, gy * PI / 180.0f, gz * PI / 180.0f, my, mx, mz);
In addition I am working in Arduino UNO.

kriswiner, 2018-02-20:
Is this order NED?

And how are you calibrating the MPU9250 sensors? Are they well-calibrated?
How do you know?

HwangHanGyeol, 2018-02-21:
What is NEO?

I did not calibrate the MPU9250 sensor.
How do I calibrate?

kriswiner, 2018-02-21:
North, East, Down is the orientation, there is also ENU. But the order has
to follow some convention.

Madgwick (N, E, D, N, E, D, N, E, D)

Also, if you don;t calibrate your sensors, especially th magnetometer, you
will no get accurate results.

See:
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

HwangHanGyeol, 2018-02-21:
I saw the link you sent.
But I do not know what to put in dest1 and dest2.
So I could not use it.

kriswiner, 2018-02-21:
Try here:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

HwangHanGyeol, 2018-02-21:
I have only MPU9250 and Arduino.
So I deleted and debug the unnecessary parts from that source.
The baud rate was also adjusted, but the characters were garbled.
Also, the value is in the middle, but the value is zero.

kriswiner, 2018-02-21:
You are going to have to fix this on your own. I've given you all I can.

HwangHanGyeol, 2018-02-21:
Ok, thanks.
I will leave a comment if I have to contact you next time.