kriswiner/MPU9250 #237: I am 0xFF I should be 0x48 error

dibs06, 2018-02-20:
I am using  Sparkfun IMU M0 which combines a MPU9250 and SMD21 microprocessor . I am using Arduino mega and I2C communication protocol for data processing. I am using your AHRS code fro sensor calibration and processong the data.
But I get some error which may be the i2c program error
I was first used I2C check and its return

Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
done

Then I was compile your AHRS code and its returns 

 MPU9250 initialized for active data mode....
AK8963 I AM 0xFF I should be 0x48
Communication failed, abort!

what should I do?
Please give me some Idea about AHRS code
Thank you

kriswiner, 2018-02-20:
No idea, did you enable pass through mode on the MPU9250?

dibs06, 2018-02-21:
How to enable pass through mode in MPU9250? I have no idea about that. 
Your cooperation is highly appreciated kris.
Thank you

kriswiner, 2018-02-21:
Did you read the data sheet and register map? Look at the sketches in this
repository?

kriswiner, 2018-02-21:
// Configure Interrupts and Bypass Enable
  // Set interrupt pin active high, push-pull, hold interrupt pin level
HIGH until interrupt cleared,
  // clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional
chips
  // can join the I2C bus and all can be controlled by the Arduino as master
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecond
pulse and any read to clear
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit
0) interrupt

kriswiner, 2018-03-21:
> @kriswiner <https://github.com/kriswiner>
> I have the same problem, my code comes from Sparkfun

Do you have this in your code?

  // Configure Interrupts and Bypass Enable
  // Set interrupt pin active high, push-pull, hold interrupt pin level
HIGH until interrupt cleared,
  // clear on read of INT_STATUS, and enable I2C_BYPASS_EN so additional
chips
  // can join the I2C bus and all can be controlled by the Arduino as master
   writeByte(MPU9250_ADDRESS, INT_PIN_CFG, 0x12);  // INT is 50 microsecond
pulse and any read to clear
   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x01);  // Enable data ready (bit
0) interrupt

naveedanwarbhatti, 2018-05-03:
@dibs06 @glucee I'm facing the same problem. Have you guys solved your problem yet?

naveedanwarbhatti, 2018-05-03:
@glucee Thanks a lot. It worked

kriswiner, 2018-05-18:
> I am a kind of new to this ..pardon me if my doubt is too silly..I am
> facing the same issue and when I add the suggested code, I am getting the
> error that 'writeByte' was not declared in this scope..what should I do?

You will need to define the writeByte method; cpy from one of the reposited
sketches? Which MCU are you using?

BhavyaKrishna, 2018-05-18:
I am kind of new to this.So pardon if my doubt is too silly.I'm facing the same issue and when I tried adding the suggested piece of code, i am getting the error 
'writeByte' was not declared in this scope
What should I do?

BhavyaKrishna, 2018-05-18:
I am using ATMEGA328 on the arduino UNO

kriswiner, 2018-05-18:
This sketch doesn;t work?

https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino

BhavyaKrishna, 2018-05-18:
I didn't try this code .I tried the MPU9250BasicAHRS_I2C.ino. I will try this and tell..Thanks for the quick reply.

BhavyaKrishna, 2018-05-18:
I am getting the error not enough memorywhen i ran that code .Do you by any chance know smaller code just to serially print the roll, pitch and yaw alone without all the lcd part?
also I tried defining a method for writebyte, but the error still persisted .

naveedanwarbhatti, 2018-05-18:
try this one: https://github.com/naveedanwarbhatti/MPU-9250-Arduino

BhavyaKrishna, 2018-05-18:
I am getting the same error again.
AK8963 I AM 0xFF I should be 0x48
Communication failed, abort!..
How exactly did you overcome the error.And did you also use UNO?

naveedanwarbhatti, 2018-05-18:
I changed the AK8963 from 0x49 to 0x00 (thanks to @glucee) and it resolved the error. 
#define WHO_AM_I_AK8963  0x49 -> #define WHO_AM_I_AK8963  0x00

And yes! I'm running it on UNO

naveedanwarbhatti, 2018-05-18:
/src/MPU9250.h line#23

BhavyaKrishna, 2018-05-18:
Thank you so much it worked..