kriswiner/MPU9250 #239: Euler Delay

beratozbay, 2018-02-24:
Hello,

I have a sparkfun MPU9250 breakout board and a teensy 3.5. When I tried the example "MPU9250BasicAHRS_t3" I take the this kind of result.

```
ax = -61.89 ay = 0.79 az = 930.85 mg
gx = -0.02 gy = 0.08 gz = 0.00 deg/s
mx = -119 my = -125 mz = 382 mG
q0 = 0.27 qx = -0.30 qy = 0.91 qz = 0.14
Yaw, Pitch, Roll: -151.28, 34.77, 174.09
Rate = 0.17 Hz
---------------------------------------------------------------------
ax = -64.51 ay = 0.49 az = 936.34 mg
gx = -0.05 gy = 0.05 gz = 0.07 deg/s
mx = -135 my = -137 mz = 367 mG
q0 = 0.33 qx = -0.28 qy = 0.89 qz = 0.15
Yaw, Pitch, Roll: -152.97, 41.85, 173.53
Rate = 6015.74 Hz
---------------------------------------------------------------------
ax = -63.96 ay = 0.55 az = 935.73 mg
gx = -0.13 gy = 0.08 gz = 0.06 deg/s
mx = -124 my = -123 mz = 374 mG
q0 = 0.39 qx = -0.27 qy = 0.86 qz = 0.17
Yaw, Pitch, Roll: -155.29, 50.34, 172.48
Rate = 6061.90 Hz
---------------------------------------------------------------------
ax = -64.82 ay = -1.53 az = 937.19 mg
gx = -0.08 gy = 0.00 gz = -0.00 deg/s
mx = -119 my = -128 mz = 389 mG
q0 = 0.47 qx = -0.24 qy = 0.83 qz = 0.19
Yaw, Pitch, Roll: -159.16, 60.73, 170.03
Rate = 6017.94 Hz
---------------------------------------------------------------------
ax = -64.27 ay = -0.31 az = 936.28 mg
gx = -0.04 gy = 0.06 gz = 0.05 deg/s
mx = -139 my = -127 mz = 377 mG
q0 = 0.56 qx = -0.21 qy = 0.77 qz = 0.21
Yaw, Pitch, Roll: -168.56, 73.20, 162.11
Rate = 6064.35 Hz
---------------------------------------------------------------------
ax = -64.76 ay = -1.10 az = 932.19 mg
gx = 0.02 gy = 0.10 gz = 0.02 deg/s
mx = -117 my = -130 mz = 395 mG
q0 = 0.67 qx = -0.17 qy = 0.68 qz = 0.24
Yaw, Pitch, Roll: 116.67, 84.48, 88.63
Rate = 6000.43 Hz
---------------------------------------------------------------------
ax = -62.19 ay = 1.10 az = 937.07 mg
gx = -0.13 gy = 0.09 gz = -0.00 deg/s
mx = -122 my = -128 mz = 369 mG
q0 = 0.78 qx = -0.13 qy = 0.56 qz = 0.27
Yaw, Pitch, Roll: 43.41, 68.89, 15.96
Rate = 6062.08 Hz
---------------------------------------------------------------------
ax = -63.96 ay = -1.34 az = 935.49 mg
gx = -0.10 gy = -0.00 gz = 0.04 deg/s
mx = -128 my = -123 mz = 377 mG
q0 = 0.88 qx = -0.08 qy = 0.36 qz = 0.30
Yaw, Pitch, Roll: 34.38, 43.28, 5.62
Rate = 6007.13 Hz
---------------------------------------------------------------------
ax = -63.84 ay = -0.49 az = 934.02 mg
gx = -0.11 gy = -0.00 gz = -0.04 deg/s
mx = -121 my = -120 mz = 377 mG
q0 = 0.93 qx = -0.06 qy = 0.10 qz = 0.36
Yaw, Pitch, Roll: 36.31, 13.50, -2.08
Rate = 6061.65 Hz
---------------------------------------------------------------------
ax = -62.19 ay = 0.06 az = 936.58 mg
gx = -0.11 gy = 0.07 gz = 0.04 deg/s
[... 270 lines cut ...]
---------------------------------------------------------------------
ax = -64.15 ay = -0.18 az = 932.86 mg
gx = -0.02 gy = 0.02 gz = 0.02 deg/s
mx = -36 my = -21 mz = 372 mG
q0 = 0.81 qx = -0.02 qy = 0.03 qz = 0.59
Yaw, Pitch, Roll: 66.22, 4.04, -0.05
Rate = 6045.68 Hz
---------------------------------------------------------------------
ax = -61.16 ay = -1.40 az = 937.19 mg
gx = 0.02 gy = 0.05 gz = 0.02 deg/s
mx = -52 my = 5 mz = 381 mG
q0 = 0.81 qx = -0.02 qy = 0.03 qz = 0.59
Yaw, Pitch, Roll: 66.54, 3.83, -0.13
Rate = 6061.86 Hz
---------------------------------------------------------------------
ax = -59.39 ay = -2.08 az = 936.83 mg
gx = -0.03 gy = 0.00 gz = 0.11 deg/s
mx = -38 my = -8 mz = 384 mG
q0 = 0.80 qx = -0.02 qy = 0.02 qz = 0.60
Yaw, Pitch, Roll: 67.35, 3.52, 0.00
Rate = 6027.97 Hz
```
As you can see there is a stabilization period at the begining. Yaw goes to 118 degree. Then I turned the board to 90 degree left and again a stabilization period until degree 66. I don't know where the problem is.

My first usage of IMU's and I will use the components for gesture recognition from wrist.

kriswiner, 2018-02-24:
How did you calibrate the sensors?

kriswiner, 2018-02-25:
> I didn't change the example code too much. My calibrate function is
>
> void calibrateMPU9250(float * dest1, float * dest2)
> {
>   uint8_t data[12]; // data array to hold accelerometer and gyro x, y, z, data
>   uint16_t ii, packet_count, fifo_count;
>   int32_t gyro_bias[3]  = {0, 0, 0}, accel_bias[3] = {0, 0, 0};
>
>  // reset device
>   writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x80); // Write a one to bit 7 reset bit; toggle reset device
>   delay(100);
>
>  // get stable time source; Auto select clock source to be PLL gyroscope reference if ready
>  // else use the internal oscillator, bits 2:0 = 001
>   writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x01);
>   writeByte(MPU9250_ADDRESS, PWR_MGMT_2, 0x00);
>   delay(200);
>
> // Configure device for bias calculation
>   writeByte(MPU9250_ADDRESS, INT_ENABLE, 0x00);   // Disable all interrupts
>   writeByte(MPU9250_ADDRESS, FIFO_EN, 0x00);      // Disable FIFO
>   writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x00);   // Turn on internal clock source
>   writeByte(MPU9250_ADDRESS, I2C_MST_CTRL, 0x00); // Disable I2C master
>   writeByte(MPU9250_ADDRESS, USER_CTRL, 0x00);    // Disable FIFO and I2C master modes
>   writeByte(MPU9250_ADDRESS, USER_CTRL, 0x0C);    // Reset FIFO and DMP
> [... 150 lines cut ...]
> Is mpu9250 not a good sensor?
> Can't my components just can't calculate it fast enough?
>
> I am reall stuck and I want to achive this speed and accuracy
> https://www.youtube.com/watch?v=BXsGWoOMtmU

You are going to have to calibrate the magnetometer too. See here
<https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration>
.

This sketch has an example.

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

beratozbay, 2018-03-05:
Sorry for my late response.

I tried different calibration tecqniues both [yours ](https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration) and [diydrones](https://diydrones.com/profiles/blogs/advanced-hard-and-soft-iron-magnetometer-calibration-for-dummies) second one worked well but not perfect.

When I calculate angles pitch behaves abnormal (I think it's abnormal). When Pitch goes exactly up suddenly yaw goes to 270(initally 90) and roll goes to -180 (initally 0).

My madgwick parameters.
`MadgwickQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f,  my,  mx, -mz);`

The sudden change I mentioned doesn't seem bad in [visualizer](https://github.com/arduino-libraries/MadgwickAHRS/tree/master/extras/Visualizer). Even so, changes not well for me because I want to use the angles as joystick angles.

kriswiner, 2018-03-05:
This is in ENU convention, and should work. Maybe the accel nd gyro aren;t
calibrated?

How do you check your sensor calibrations?

beratozbay, 2018-03-05:
For accel and gyro, I just check the outputs while sensor standing.
For mag, I wrote a program in processing. It has output mag points in 2D(Mxy, Mxz, Myz).

kriswiner, 2018-03-05:
Not sure what to tell you. Although the Sparkfun design has power traces
running under the MPU9250, maybe tthese are interfering with your
measurements...

beratozbay, 2018-03-07:
I see. Thank you so much. I learned a lot while trying to calibrate sensor.
But I want to ask an out of topic question.

Can you suggest a high quality sensor(s) for motion tracking?

kriswiner, 2018-03-07:
What kind of motion tracking? Absolute orientation, dead reckoning,
velocimetry...?

beratozbay, 2018-03-07:
Motion recognition. Like smart watches you know

kriswiner, 2018-03-07:
You mean running versus walking versus sitting versus standing?

Can you be more specific?

beratozbay, 2018-03-07:
Yes I want to able to recognize mainly runnig, walking, sitting, standig. Besides I will try to recognize specific movements like tennis or fitness movement.

kriswiner, 2018-03-07:
This will be more about the contextual interpretation of the data then than
the quality of the sensor or sensor fusion. I would use either 1) an
MPU9250 or 2) an MPU9250 + EM7180. There are also some sensors that have
contextual awareness libraries that are offered as binaries like the
LSM6DSM, which is a fine sensor.