kriswiner/MPU9250 #240: Magnetometer resolution change

cero10, 2018-02-28:
Hello, first of all i want to thank you for your project they are really useful to me.

I am using your MPU9250BasicAHRS.ino project, i would like to know the exact lines i should use to change mpu9250 magnetometer's sample rate of to 100hz, if im right that is the 16bit resolution..
And what function should i call to know at what resolution the accel and gyro are working.

kriswiner, 2018-02-28:
uint8_t Mscale = MFS_16BITS; // Choose either 14-bit or 16-bit magnetometer
resolution
uint8_t Mmode = 0x02; // 2 for 8 Hz, 6 for 100 Hz continuous magnetometer
data read

cero10, 2018-02-28:
I do not understand why is this happening, i got a " 'MFS_16BITS' was not declared in this scope" error message. Isn't **uint8_t Mscale = MFS_16BITS;** a declaration?

kriswiner, 2018-02-28:
I thought you said you were using the above sketch? Those lines are in the
sketch...

cero10, 2018-02-28:
I am using MPU9250BasicAHRS.ino and i do not see them anywhere. Should i use the MPU9250BasicAHRS_t3.ino that contain those lines?? because i am using the non-t3 version..

cero10, 2018-02-28:
can you point me to your i2c_t3.h library? i cant find it and there are various declaration problems when running your sketch without your i2c lib.

kriswiner, 2018-02-28:
Teensy forum

cero10, 2018-02-28:
Is this sketch only for teensy?

cero10, 2018-02-28:
MPU9250BasicAHRS_t3.ino

kriswiner, 2018-02-28:
MPU9250BasicAHRS_*t3*.ino

requires Teensy and i2c_t3.h wire library.

cero10, 2018-02-28:
So i should use the MPU9250BasicAHRS.ino?
if yes, how to change res and sample rate for arduino?

kriswiner, 2018-02-28:
I already told you

cero10, 2018-02-28:
So lets summarize, i am using the MPU9250BasicAHRS.ino (which does not contain those 2 lines)

>  uint8_t Mscale = MFS_16BITS;

>  uint8_t Mmode = 0x02;

and i still get the "was not declared in this scope" error **ONLY** for the Mscale=MFS_16BITS. So can you help me or is there something else i am not doing right?

kriswiner, 2018-02-28:
Help me out here,  can you tell me what is on lines 226/227 of this sketch:

https://github.com/kriswiner/MPU9250/blob/master/MPU9250BasicAHRS.ino

and if you try to compile this, are you saying you get an error?

I would try this sketch anyway:

https://github.com/kriswiner/MPU9250/tree/master/MPU9250_
BME280_SPIFlash_Ladybug

the one you are using is very old. But you will have to  make some changes
unless you are using a Ladybug dev board.

kriswiner, 2018-02-28:
I think the problem is you need to add the quaternionFilters.ino file to
your sketch tab.

cero10, 2018-02-28:
Yep it works now. so i got some problems, the q0,qx,qy,qz are constantly {1.0.0.0} so im guessing it has something to do with the MahonyQuaternionUpdate(..);?? and by default it's using the mahony filter, what are the advantages over Madgwich

kriswiner, 2018-02-28:
You need to calibrate your sensors. I would switch to a different sketch.

cero10, 2018-02-28:
ok then, one last question. Is there a possibility that i get a pitch that doesn't get affected by high dynamics with either Mahony or Madgwick filter?

kriswiner, 2018-02-28:
I don;t understand the question, what does high dynamics mean?

cero10, 2018-02-28:
I mean that when you rotate the sensor slowly it shows you accurately the pitch and roll but when you make sudden movements the accel values are all over the place so the pitch/roll are wrong too. I am literally asking if the Mahony or Madgwick filter will almost zero that error by throwing gyro and mag values into the formula?

kriswiner, 2018-02-28:
Should do

cero10, 2018-02-28:
ok then thank you very much. you were really helpful man!!