kriswiner/MPU9250 #241: please evaluate my calibration result of MPU 9250

exzibith, 2018-03-01:
I'm using WEMOS Lolin32 board and merge your MPU9250_MS5637_AHRS_t3.ino code.
calibration results as below :

MPU9250
9-DOF 16-bit
motion sensor
60 ug LSB
Scanning...
I2C device found at address 0x0C  !
I2C device found at address 0x68  !
done

MPU9250 9-axis motion sensor...
MPU9250 I AM 73 I should be 73
MPU9250 is online...
x-axis self test: acceleration trim within : -0.4% of factory value
y-axis self test: acceleration trim within : 0.1% of factory value
z-axis self test: acceleration trim within : 0.6% of factory value
x-axis self test: gyration trim within : -0.5% of factory value
y-axis self test: gyration trim within : -0.3% of factory value
z-axis self test: gyration trim within : 0.3% of factory value
 Calibrate gyro and accel
accel biases (mg)
11.11
45.23
-17.58
gyro biases (dps)
2.89
0.05
-0.92
MPU9250 bias
 x   y   z  
1145-17mg2.90.0-0.9o/s
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
Mag Calibration: Wave device in a figure eight until done!
Mag Calibration done!
AK8963 mag biases (mG)
-253.08
709.87
-765.09
AK8963 mag scale (mG)
1.87
0.49
2.47
X-Axis sensitivity adjustment value 1.16
Y-Axis sensitivity adjustment value 1.17
Z-Axis sensitivity adjustment value 1.13
AK8963
ASAX 
1.16
ASAY 
1.17
ASAZ 
1.13
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = 0 my = 0 mz = 0 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 0.42, 0.00, 0.00
Grav_x, Grav_y, Grav_z: 0.00, 0.00, 1000.00 mg
Lin_ax, Lin_ay, Lin_az: 0.00, 0.00, -1000.00 mg
rate = 1.02 Hz

 x   y   z : 0, 0, 0 mg
0, 0, 0 o/s
0, 0, 0 mG
0, 0, 0 ypr
ax = 88.93 ay = 360.78 az = 980.59 mg
gx = 3.34 gy = 2.61 gz = -93.29 deg/s
mx = 13 my = 55 mz = -20 mG
q0 = 0.99 qx = -0.00 qy = 0.00 qz = 0.12
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 13.66, 0.48, -0.11
Grav_x, Grav_y, Grav_z: 1.87, 8.31, 999.96 mg
Lin_ax, Lin_ay, Lin_az: 87.06, 352.47, -19.37 mg
rate = 25228.31 Hz

 x   y   z : 88, 360, 980 mg
3, 2, -93 o/s
13, 55, -20 mG
13, 0, 0 ypr
ax = -126.46 ay = -424.93 az = 964.42 mg
gx = 9.85 gy = -15.11 gz = 78.01 deg/s
mx = -3 my = 51 mz = -16 mG
q0 = 0.99 qx = 0.00 qy = -0.02 qz = 0.12
Gyro temperature is 29.0 degrees C
Yaw, Pitch, Roll: 14.19, -1.91, -0.15
Grav_x, Grav_y, Grav_z: 2.61, -33.32, 999.44 mg
Lin_ax, Lin_ay, Lin_az: -129.08, -391.60, -35.02 mg
rate = 24958.09 Hz

 x   y   z : -126, -424, 964 mg
9, -15, 78 o/s
-3, 51, -16 mG
14, -1, 0 ypr
ax = -206.60 ay = 250.85 az = 972.05 mg
gx = 2.91 gy = 3.18 gz = 40.81 deg/s
mx = -55 my = -13 mz = -8 mG
q0 = 1.00 qx = -0.01 qy = -0.02 qz = -0.08
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 350.83, -1.96, -0.76
Grav_x, Grav_y, Grav_z: 13.20, -34.25, 999.33 mg
Lin_ax, Lin_ay, Lin_az: -219.81, 285.10, -27.28 mg
rate = 25138.67 Hz

 x   y   z : -206, 250, 972 mg
2, 3, 40 o/s
-55, -13, -8 mG
350, -1, 0 ypr
ax = 283.08 ay = -431.52 az = 923.46 mg
gx = -3.99 gy = -13.04 gz = -50.22 deg/s
mx = -39 my = -12 mz = -62 mG
q0 = 0.99 qx = 0.01 qy = 0.00 qz = -0.12
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 346.40, 0.20, 0.88
Grav_x, Grav_y, Grav_z: -15.37, 3.44, 999.88 mg
Lin_ax, Lin_ay, Lin_az: 298.45, -434.96, -76.41 mg
rate = 24923.69 Hz

 x   y   z : 283, -431, 923 mg
-3, -13, -50 o/s
-39, -12, -62 mG
346, 0, 0 ypr
ax = -76.17 ay = 44.74 az = 1054.02 mg
gx = -5.98 gy = 4.36 gz = -4.61 deg/s
mx = -32 my = 16 mz = 37 mG
q0 = 1.00 qx = -0.02 qy = -0.02 qz = -0.04
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 355.60, -2.30, -2.68
Grav_x, Grav_y, Grav_z: 46.65, -40.14, 998.10 mg
Lin_ax, Lin_ay, Lin_az: -122.82, 84.88, 55.91 mg
rate = 25006.99 Hz

 x   y   z : -76, 44, 1054 mg
-5, 4, -4 o/s
-32, 16, 37 mG
355, -2, -2 ypr
ax = -590.39 ay = 7.08 az = 1024.84 mg
gx = 5.65 gy = 6.39 gz = 21.10 deg/s
mx = 26 my = 82 mz = -20 mG
q0 = 0.99 qx = -0.00 qy = -0.02 qz = 0.12
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 14.49, -1.73, -0.29
Grav_x, Grav_y, Grav_z: 5.01, -30.28, 999.53 mg
Lin_ax, Lin_ay, Lin_az: -595.40, 37.36, 25.31 mg
rate = 25099.67 Hz

 x   y   z : -590, 7, 1024 mg
5, 6, 21 o/s
26, 82, -20 mG
14, -1, 0 ypr
ax = 91.80 ay = -119.75 az = 1031.62 mg
gx = 6.31 gy = -3.78 gz = 45.30 deg/s
mx = -65 my = 11 mz = -20 mG
q0 = 0.99 qx = -0.01 qy = -0.03 qz = -0.10
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 349.14, -3.36, -1.34
Grav_x, Grav_y, Grav_z: 23.39, -58.66, 998.00 mg
Lin_ax, Lin_ay, Lin_az: 68.40, -61.09, 33.61 mg
rate = 25092.80 Hz

 x   y   z : 91, -119, 1031 mg
6, -3, 45 o/s
-65, 11, -20 mG
349, -3, -1 ypr
ax = 58.11 ay = 272.52 az = 1021.00 mg
gx = -2.88 gy = -0.66 gz = 70.86 deg/s
mx = -42 my = -46 mz = -16 mG
q0 = 0.97 qx = -0.01 qy = -0.03 qz = -0.25
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 331.69, -3.58, -0.78
Grav_x, Grav_y, Grav_z: 13.55, -62.41, 997.96 mg
Lin_ax, Lin_ay, Lin_az: 44.56, 334.93, 23.04 mg
rate = 24969.78 Hz

 x   y   z : 58, 272, 1020 mg
-2, 0, 70 o/s
-42, -46, -16 mG
331, -3, 0 ypr
ax = -81.54 ay = -174.32 az = 1027.95 mg
gx = -8.19 gy = -1.12 gz = -53.87 deg/s
mx = -26 my = -9 mz = -62 mG
q0 = 0.98 qx = -0.02 qy = -0.02 qz = -0.19
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 338.70, -3.14, -1.53
Grav_x, Grav_y, Grav_z: 26.71, -54.79, 998.14 mg
Lin_ax, Lin_ay, Lin_az: -108.25, -119.53, 29.81 mg
rate = 25058.47 Hz

 x   y   z : -81, -174, 1027 mg
-8, -1, -53 o/s
-26, -9, -62 mG
338, -3, -1 ypr
ax = -32.35 ay = -10.80 az = 989.07 mg
gx = 0.49 gy = -0.05 gz = 1.79 deg/s
mx = -58 my = 21 mz = -54 mG
q0 = 0.99 qx = -0.02 qy = -0.03 qz = -0.13
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 345.72, -4.06, -1.72
Grav_x, Grav_y, Grav_z: 29.94, -70.87, 997.04 mg
Lin_ax, Lin_ay, Lin_az: -62.29, 60.06, -7.96 mg
rate = 25147.32 Hz

 x   y   z : -32, -10, 989 mg
0, 0, 1 o/s
-58, 21, -54 mG
345, -4, -1 ypr
ax = -19.29 ay = 8.18 az = 995.61 mg
gx = -0.05 gy = 0.12 gz = -1.88 deg/s
mx = -55 my = 22 mz = -41 mG
q0 = 0.99 qx = -0.03 qy = -0.03 qz = -0.15
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 342.70, -4.27, -2.68
Grav_x, Grav_y, Grav_z: 46.66, -74.43, 996.13 mg
Lin_ax, Lin_ay, Lin_az: -65.95, 82.61, -0.53 mg
rate = 25316.27 Hz

 x   y   z : -19, 8, 995 mg
0, 0, -1 o/s
-55, 22, -41 mG
342, -4, -2 ypr
ax = -7.51 ay = 10.31 az = 992.25 mg
gx = 0.19 gy = 0.18 gz = -0.10 deg/s
mx = -29 my = 22 mz = -25 mG
q0 = 0.98 qx = -0.03 qy = -0.03 qz = -0.17
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 340.52, -4.24, -3.19
Grav_x, Grav_y, Grav_z: 55.56, -73.89, 995.72 mg
Lin_ax, Lin_ay, Lin_az: -63.06, 84.20, -3.47 mg
rate = 25437.74 Hz

 x   y   z : -7, 10, 992 mg
0, 0, 0 o/s
-29, 22, -25 mG
340, -4, -3 ypr
ax = 89.72 ay = 19.47 az = 1062.68 mg
gx = -9.94 gy = -2.55 gz = -48.47 deg/s
mx = -6 my = 40 mz = -12 mG
q0 = 0.99 qx = -0.04 qy = -0.01 qz = -0.14
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 344.50, -2.34, -4.67
Grav_x, Grav_y, Grav_z: 81.33, -40.77, 995.85 mg
Lin_ax, Lin_ay, Lin_az: 8.39, 60.24, 66.83 mg
rate = 25083.33 Hz

 x   y   z : 89, 19, 1062 mg
-9, -2, -48 o/s
-6, 40, -12 mG
344, -2, -4 ypr
ax = -320.56 ay = -130.19 az = 869.93 mg
gx = 9.10 gy = 9.52 gz = 35.00 deg/s
mx = -16 my = 54 mz = 8 mG
q0 = 1.00 qx = -0.02 qy = -0.02 qz = -0.07
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 352.77, -2.76, -2.66
Grav_x, Grav_y, Grav_z: 46.40, -48.08, 997.77 mg
Lin_ax, Lin_ay, Lin_az: -366.95, -82.11, -127.83 mg
rate = 25107.07 Hz

 x   y   z : -320, -130, 869 mg
9, 9, 34 o/s
-16, 54, 8 mG
352, -2, -2 ypr
ax = 39.49 ay = 119.75 az = 1056.58 mg
gx = -15.62 gy = 1.25 gz = 18.19 deg/s
mx = -68 my = 15 mz = 16 mG
q0 = 0.97 qx = -0.04 qy = -0.02 qz = -0.22
Gyro temperature is 29.1 degrees C
Yaw, Pitch, Roll: 335.43, -3.75, -4.06
Grav_x, Grav_y, Grav_z: 70.58, -65.33, 995.36 mg
Lin_ax, Lin_ay, Lin_az: -31.09, 185.09, 61.22 mg
rate = 25129.30 Hz

 x   y   z : 39, 119, 1056 mg
-15, 1, 18 o/s
-68, 15, 16 mG
335, -3, -4 ypr
====================================================
====================================================

1. Is this calibration result correct?
2. how to plot the calibration data results? is it necessary third party software?
3. does each board restart need to calibrate the data with eight figure?
4. I'm trying to create a project like the freedrum stick website (freedrum.rock), does this sketch match my project, to calibrate the MPU-9250?
5. i'm in need of data results from mpu 9250 which is already converted to degrees and angles 0-360 degrees?
6. I want every time I restart the wemos board wherever the direction of its position, I get the data value of 0 degree angle.

thanks a lot for the help and input, I'm still new to using MPU-9250, so still have to ask a lot of the experts.

kriswiner, 2018-03-01:
Something is wrong with our gyro data.

Calibration is a function of the specific sensor youa re using. Once you
have good calibration data, you can store in the sketch and not have to
calibrate again, or for a while anyway.

kriswiner, 2018-03-01:
wrong with your gyro data.

exzibith, 2018-03-01:
which one is wrong with the data gyro? Is this?
I tested it with figure 8,

gx = 9.85 gy = -15.11 gz = 78.01 deg/s
gx = 2.91 gy = 3.18 gz = 40.81 deg/s

whether for the next data mag bias calibration can i use my own way?
as follows?

readMagData (magCount); // Read the x / y / z adc values
    getMres (); // Supplement
     magBias [0] = -237.37; // Additional calibration results User environmental x-axis correction in milliGauss, should be automatically calculated
     magBias [1] = +771.37; // Additional calibration results User environmental x-axis correction in milliGauss
     magBias [2] = -775.24; // Additional calibration results User environmental x-axis correction in milliGauss

kriswiner, 2018-03-01:
Gyro at rest (no motion) should be much less that 1 dps.

Yes, once you have the calibration right, you can embed in the sketch.