kriswiner/MPU9250 #243: Changing the sensitivity of the gyroscope

MitalPattani, 2018-03-04:
I wish to reduce the sensitivity of the gyroscope, what are the commands?

kriswiner, 2018-03-04:
datasheet

MitalPattani, 2018-03-08:
how do i stabilize the gyroscope readings ??
I am getting these reading for a 9250 fixed in a stationary position....!

Yaw, Pitch, Roll: 141.22, 16.24, 25.58
Yaw, Pitch, Roll: -125.38, -44.45, -16.70
Yaw, Pitch, Roll: -15.73, 56.26, -86.41
Yaw, Pitch, Roll: -108.90, -53.67, 171.62
Yaw, Pitch, Roll: 149.51, 32.95, -179.43
Yaw, Pitch, Roll: 139.22, -26.22, -157.63
Yaw, Pitch, Roll: 105.41, 24.71, 151.47
Yaw, Pitch, Roll: 111.34, -16.30, -149.41
Yaw, Pitch, Roll: 102.08, 9.19, 125.80
Yaw, Pitch, Roll: 82.28, -20.45, -159.66
Yaw, Pitch, Roll: 132.97, 16.90, 102.85
Yaw, Pitch, Roll: 76.35, -51.32, 170.40
Yaw, Pitch, Roll: -176.76, 41.77, 130.74
Yaw, Pitch, Roll: 111.64, -39.56, 169.71
Yaw, Pitch, Roll: 170.84, 40.94, 123.98
Yaw, Pitch, Roll: 108.37, -41.38, 167.67

kriswiner, 2018-03-08:
Did you try calibrating the sensors?

MitalPattani, 2018-03-09:
I've tried the function imu.calibrateMPU9250(accelBias, gyroBias)
And this is the result!

kriswiner, 2018-03-09:
Might need to calibrate the magnetometer...

MitalPattani, 2018-03-09:
[calibrate.txt](https://github.com/kriswiner/MPU9250/files/1798015/calibrate.txt)
I have done that too in this code and it still doesn't seem to have calibrated.
I added another small code for averaging the values of 'g' and 'a' for 1000 values and subtracting it from each reading. Will this work?? 
Sorry for troubling you so much and thank you so much for the quick responses!!!!

kriswiner, 2018-03-09:
" Will this work?? "

No idea, there are plenty of examples of calibration functions that do in
my github repository.

MitalPattani, 2018-03-13:
Thanks a ton!🍻

Abhi26021289, 2021-03-19:
Hi Kris, I am working on a very simple device that plays music when you lift the box up. and it stops the music when you place the device back on the flat surface. the problem is I am not able to calibrate the sensor and using this code I am just able to play the music but it's not behaving the way I want it to.
please help me.
and i have zero understanding of the coding language.
#include "MPU9250.h"
#include "DYPlayerArduino.h"
#include <SoftwareSerial.h>

MPU9250 mpu;
//int buttonPin = 2;
volatile int state = LOW;
//int buttonState = 0;

SoftwareSerial SoftSerial(10, 11);
DY::Player player(&SoftSerial);

boolean play_flg = false;

boolean calibration_flg = false;
int yaw_val = 0;
int cal_count = 0;
void setup() {

  player.begin();
  player.setVolume(30);
  player.stop();

  //pinMode(buttonPin, INPUT_PULLUP);
  Serial.begin(9600);
  Wire.begin();
  delay(2000);

  if (!mpu.setup(0x68)) {  // change to your own address
    while (1) {
      Serial.println("MPU connection failed. Please check your connection with `connection_check` example.");
      delay(5000);
    }
  }

}

void loop() {
  if (mpu.update()) {
    static uint32_t prev_ms = millis();
    if (millis() > prev_ms + 25) {
      print_roll_pitch_yaw();
      prev_ms = millis();
    }
  }

  //player.playSpecified(2);
  //delay(2000);
}

void print_roll_pitch_yaw() {
  /*
    Serial.print("Yaw, Pitch, Roll: ");
    Serial.print(mpu.getYaw(), 2);
    Serial.print(", ");
    Serial.print(mpu.getPitch(), 2);
    Serial.print(", ");
    Serial.println(mpu.getRoll(), 2);
  */

  if (!calibration_flg) {
    Serial.print("Calibration: ");
    yaw_val = mpu.getYaw();
    //Serial.print(yaw_val);
    if (yaw_val > -100 || yaw_val < 30 ) {
      cal_count++;
      delay(100);
      if (cal_count > 30) {
        calibration_flg = true;
        Serial.println("START");
        cal_count = 0;
      }
    }
    Serial.println(mpu.getYaw());

  }

if (calibration_flg) {
    if ((mpu.getYaw() < -2) || (mpu.getYaw() > 2)) {
      Serial.print("pressed: ");
      Serial.print(mpu.getYaw());
      Serial.println();

      if (play_flg == false) {
        play_flg = true;
        Serial.println("Audio Play");
        audio_play();
      } else {
        play_flg = false;
        Serial.println("No Audio");
        player.pause();
        //audio_pause();
      }
    }
  }
}

void audio_play() {
  player.playSpecified(2);
  delay(19000);
  play_flg = false;
}

kriswiner, 2021-03-19:
I cannot debug your code for you...