kriswiner/MPU9250 #244: Calibration questions

RicardPalo, 2018-03-06:
Hello Kris,
I'm finishing my university final project, which is an AGV guided by GPS. I need to get the yaw in order to have a reference point for AGV's moves. I'm using your sketchbooks, but I have many questions:
a) Using your code (with some changes) I think the yaw gotten is properly. But, when I check the Mz, Mx and My plots, it doesn't look well calibrated. How can it be? I don't know what I'm doing wrong with the calibration. My plot: 
[image]

b) My second questions is about magScale. I don't understand why it is used for? When we call for the function I use this:  [magcalMPU9250(magBias,magScale);] 
` 
void magcalMPU9250(float * dest1, float * dest2) 
 {

 uint16_t ii = 0, sample_count = 0;
 int32_t mag_bias[3] = {0, 0, 0}, mag_scale[3] = {0, 0, 0};
 int16_t mag_max[3] = {-32767, -32767, -32767}, mag_min[3] = {32767, 32767, 32767}, mag_temp[3] = {0, 0, 0}; 

 Serial.println("Mag Calibration: Wave device in a figure eight until done!");

 sample_count = 128;
 for(ii = 0; ii < sample_count; ii++) {
 readMagData(mag_temp);  // Read the mag data   
 for (int jj = 0; jj < 3; jj++) {
  if(mag_temp[jj] > mag_max[jj]) mag_max[jj] = mag_temp[jj];
  if(mag_temp[jj] < mag_min[jj]) mag_min[jj] = mag_temp[jj];

 }
 delay(135);  // at 8 Hz ODR, new mag data is available every 125 ms
 }

// Get hard iron correction
 mag_bias[0]  = (mag_max[0] + mag_min[0])/2;  // get average x mag bias in counts
 mag_bias[1]  = (mag_max[1] + mag_min[1])/2;  // get average y mag bias in counts
 mag_bias[2]  = (mag_max[2] + mag_min[2])/2;  // get average z mag bias in counts

 dest1[0] = (float) mag_bias[0]*mRes*magCalibration[0];  // save mag biases in G for main program
 dest1[1] = (float) mag_bias[1]*mRes*magCalibration[1];   
 dest1[2] = (float) mag_bias[2]*mRes*magCalibration[2];  

// Get soft iron correction estimate
 mag_scale[0]  = (mag_max[0] - mag_min[0])/2;  // get average x axis max chord length in counts
 mag_scale[1]  = (mag_max[1] - mag_min[1])/2;  // get average y axis max chord length in counts
 mag_scale[2]  = (mag_max[2] - mag_min[2])/2;  // get average z axis max chord length in counts
 float avg_rad = mag_scale[0] + mag_scale[1] + mag_scale[2];

 avg_rad /= 3.0;
 dest2[0] = avg_rad/((float)mag_scale[0]);
 dest2[1] = avg_rad/((float)mag_scale[1]);
 dest2[2] = avg_rad/((float)mag_scale[2]);

 Serial.println("Mag Calibration done!");
 }   
'
c)For my project I need to make a function to get the yaw when is necessary, but I can't calibrate the magne each time I run the function. So first I calibrate the magne and take the magBiases. Finally I avoid the magne calibration and put the magBiases in the function which calculates the mx, my and mz. The problem is that it doesn't give a 0 to 360º, it goes from 80 to 250º approximately. Is this problem related to the calibrations issues?

Thank you sooo much, I know it's a long post...

kriswiner, 2018-03-07:
In order to get proper mag calibration you have to move the sensor in all
deisctions in 3D space, to sample the response surface at every point in
other words. Are you doing this?

RicardPalo, 2018-03-07:
Thank for your answer. Now I think the calibration is quite well:
[image]
So, if I use the magBiases gotten from this calibration, can get the same yaw without calibration, can't I?

Apart from this, I still don't understand why we put magScale in the magcalMPU9250(magBias,magScale); I've checked the code and magScale isn't used in other functions. Could you help me?

kriswiner, 2018-03-07:
I would say this is better but still not great. You could increase the time
for the calibration and try to move the sensor to cover more of the
response surface.

Magscale could be used to improve the deta quality and correct for cross
axis distorions and it should be used if you want the best orientation
solution this simple calibration method can provide.

RicardPalo, 2018-03-07:
I've used the magBiases gotten from this calibration, and I get good enough yaw results.  

But, if I want to increase the time, do I have to increase the "sample_count" of the "magcalMPU9250(magBias,magScale)" function? Now I have: sample_count = 128;. If I increase this number, do I have to change anything else? Which number do you recommend to put?

Thank you Kris!

kriswiner, 2018-03-07:
Try 256

RicardPalo, 2018-03-07:
I have made two tests with sample_count = 256. These are the plots:
1º
[image]
2º
[image]

I think there is not an important improve, don't you?

Anyways, if the yaw I get using this biases is enough good for my project there is no need to try to improve it more. 
Please tell me what you think.

Thank you for your time!

RicardPalo, 2018-03-14:
Hi Kris,
I would like you to answer the last question I asked. I'm a bit lost. 

Apart from that, I'm trying to use a GPS with the MPU9250. The GPS will give me the coordinates, then using this coordinates the bearing is calculated. Finally, the bearing is compared with the yaw obtained and the robot turns to the correct orientation. The problem is that when I put the GPS program with the MPU9250 program, the GPS is not detected.

My question is, the MPU definitions on the program can affect the GPS??

Please tell me what you think.
Thank you for your time!

kriswiner, 2018-03-14:
" MPU definitions on the program can affect the GPS?? "

Depends on your code, are you over-writing a bunch of variables. In
general, no.