kriswiner/MPU9250 #245: Magnetometer calibration--can I do without it?

usinjin, 2018-03-09:
Hey Kris,

I have a somewhat challenging project: I'm using a 9DOF (FXOS8700CQ + FXAS12002 accel., mag., gyro) to measure the slope of the ground in a 4' radius. The sensor is mounted at the end of a metal (non-ferrous!) "arm" that rotates 360 degrees, measuring the pitch of the sensor every ~3 degrees. I'm running a sensor fusion algorithm using Mahony. I set the gyro trims at the beginning of the run (based on average values) and am using MotionCal to get a hard- and soft-iron matrix for the magnetometer.

When the magnetometer is calibrated, the algorithm works pretty well. When the calibration is somewhat off, the drift becomes absolutely horrendous. Here's the issue: when I have to take the model to demonstrate, it's in a completely different environment (usually inside an office). I don't usually have the option to recalibrate there. The values the arm reports are very far off because of this (boy, does it not look good in front of my project manager!).

Here's my question. Besides the obvious recalibration, is there anything I can do to avoid this issue? Note that we don't need to return the yaw from the algorithm--just the roll (which is the right direction for the orientation on our arm.) 

The arm will hopefully one day be a plug-and-play solution, and at the moment, the question of an extra manual recalibration step is out of the question. Is there a quick-and-dirty sort of calibration I can do using only raw sensor values right at the start of the running routine?

kriswiner, 2018-03-09:
Use 6 DoF fusion and forget about the mag. pitch and roll only need an
accel and gyro. Madgwick has a very nice 6 DoF fusion filter for the
MPU6050 which should work well here.

https://github.com/kriswiner/MPU6050/blob/master/quaternionFilter.ino

usinjin, 2018-03-09:
Nice. Will try this!

closed 2018-03-09 by usinjin