kriswiner/MPU9250 #246: Can't connect to MPU9250

Jan-Jelle, 2018-03-12:
Good afternoon Kris,

First of all I'd like to thank you for all the work you put into this library. it helped me a great deal to get my project going! 

I started to re-write my arduino sketch to include your libraries as files instead of having all the code in the sketch itself. after changing all the function names and adding the necessary variables etc. it seems to compile fine, but i can't connect to the sensor anymore. an I2C scan shows it has address 0x68 so it should be OK. and running an example you made also works fine, but with the included libraries as a header file it seems like it doesn't read the WHOAMI register properly (0xFF comes back). after going through your header and cpp file the only difference I could find was in the read_Byte function. in the working example you've put:

	Wire.requestFrom(address, (uint8_t) 1);  // Read one byte from slave register address 

And in the CPP file it says:

Wire.requestFrom(address, 1);  // Read two bytes from slave register address 

I wondered if this extra byte could give an issue in the buffer and it doesn't read the WHOAMI register value properly. but after changing this in the CPP file to the line above it didn't help either.

I know this sounds a bit fuzzy but i've been staring at this for too long now. perhaps you can help me with this otherwise i'll try to rephrase my question tomorrow.

Thanks!

kriswiner, 2018-03-12:
Are you sure you are using 0x68 as the MPU9250 address?

Jan-Jelle, 2018-03-13:
I've taken out the #if #endif construction for the possible address of 0x69. the header file now contains:

// Define I2C addresses of MPU9250

#define MPU9250_ADDRESS 0x68   // Device address when ADO = 0
#define AK8963_ADDRESS  0x0C   //  Address of magnetometer

The GetMPU9250ID function should use this address and the WHOAMI 0x75 register as far as i can see. 
did i forget to include some function to define the address in the arduino sketch? i havn't done anything in the sketch regarding addressing..