kriswiner/MPU9250 #247: MPU9250 will be placed vertically

Jan-Jelle, 2018-03-17:
Hi Kris,

I was wondering if you could tell me how to set the offsets if i tilt the sensor 90 degrees down. I will mount the thing vertical in an enclosed box which will go subsea. this position needs to be "horizontal". is this even possible? i can set my offsets at the surface as well but since Eulers angles start acting funny at those angles my measurement are off.

Much appreciated!

kriswiner, 2018-03-17:
You will need to change the NED or NEU feed to the Madgwick filter
accordingly, and calibrate in the vertical position but this is all
straightforward.

Jan-Jelle, 2018-03-17:
Hi Kris,

Thanks for the reply. I need to change the pitch in both 90 degrees down and rotate it 90 degrees. I thought the key was in the registers 19 to 24 which are gyro offsets, but these i don't have to change then?

Verzonden vanaf mijn Samsung Galaxy-smartphone.

kriswiner, 2018-03-17:
Gyro offset have to be measured.

Jan-Jelle, 2018-03-17:
Could you help me out with this in the call for the quaternions function? I am not an experienced programmer, maybe you can point me in the right direction. What do I need to change in this part?

  MahonyQuaternionUpdate(ax, ay, az, gx*PI/180.0f, gy*PI/180.0f, gz*PI/180.0f, my, mx, mz);

kriswiner, 2018-03-17:
These functions expect the sensor data in come conventionl form, like NED
or ENU, so it has to be consistent for your application. If mounted
vertically, then Down will likely be the x- or y- axis instead of the
z-axis, etc.

Jan-Jelle, 2018-03-17:
So just swapping location correspondingly in the mahoney function call’s arguments? And does this only count for the gyro values, or all values?

I’m only reading pitch and roll with this sensor.

kriswiner, 2018-03-17:
> issuecomment-373940267

Consisten NED or ENU for all sensors.

Jan-Jelle, 2018-03-17:
Thanks a bunch 😊 had to try a few posibilities but I got it all working smoothly. Much appreciated!

DatNotData, 2020-08-13:
@Jan-Jelle 
Have you managed to get a successful "rearrangement" of the x, y, z when you input it into the MahonyQuaternionUpdate function?
I've tried a couple of  "combinations" and haven't been successful yet.

I've also tried to play with the part where the gravity is removed from the accelerometer bias calculation to be on the right axis.

kriswiner, 2020-08-13:
Try NED convention, this should work for all fusion algorithms...

DatNotData, 2020-08-17:
I tried, but the gravity component is removed from the z-axis only. Do I need to change that to the Y-axis (my new vertical axis).
Are there also other things like that that needs to be changed also?

kriswiner, 2020-08-17:
I don't understand your comment.

The Madgwick or Mahoney fusion algorithms expect data in the form of NED,
aN, aE, aD, gN, gE, gD, mN, mE, mD. select which accel axis you want to
point North and then everything else should be determined. For the MPU9250,
accel and gyro axes are the same, but not the same as mag axes...

DatNotData, 2020-08-17:
I understand that I need to input data in the form of NED, but would I need to change lines such as this one also:

    `if(accel_bias[2] > 0L) {accel_bias[2] -= (int32_t) accelsensitivity;}  // Remove gravity from the z-axis accelerometer bias calculation
    else {accel_bias[2] += (int32_t) accelsensitivity;}`

kriswiner, 2020-08-17:
No. Not for the fusion algorithm.

But yes, if you are going to do the accel calibration with the board other
than flat on the table.