kriswiner/MPU9250 #249: Custom PCB w/ MPU9250 possible trace connection issue

dirksavage88, 2018-03-24:
Hello, 

I am doing a school project that requires custom PCB circuitry, i'm using an atmega328 connected to a 9250 through two level shifting mosfets. I botched the MPU connections (I think) and accidently routed Ad0 to SDA instead of ground. It's too late to send the board back as I just received it. I'm not sure if I should just print a new board with the corrected connection, or if I can go with what I have.

If Ad0 is connected to the SDA line, will it still be found by the atmega through 0x69? Or is the SDA line in transition?

kriswiner, 2018-03-24:
Cut the trace from ADO to SDA. It won't work with these two connected.

dirksavage88, 2018-03-24:
If I cut the trace, won't ADO be floating? Does it need to be grounded? Right now it is only connected to SDA

kriswiner, 2018-03-24:
If it is floating it usually settle at one of either HIGH or LOW. This is
much better than having it connected to SDA. Ground it if you can but with
ADO floating at least the board will work.

dirksavage88, 2018-03-25:
Thanks Kris, 

I have an updated eagle cad schematic for the new revision. It's my first time working with this IMU, I would like to know if the bypass capacitors are correct:
[image]

kriswiner, 2018-03-25:
No, you need one 100 nF bypass on 8 and one on 13. And no 10 K pullup on
nCS, connect to 3V3 directly. Connect pin 1 directly to 3V3. No metal or
solder on pad. No copper on any layer and no trace on any layer directly
beneath the MPU9250 package, no exceptions. Pins 18 and 20 are ground. Make
sure you have FSYNC to GND if you're not using it. Why are you not using
the interrupt?

dirksavage88, 2018-03-25:
We are using this IMU to passively read values on a large scale quadcopter. It's going to store those values on a micro sd card reader. The data is analyzed post flight. Since our projected flight is less than a minute long, I figured we would just let the sensor run the data to the sd card continuously, as long as power is applied to the circuit.

kriswiner, 2018-03-25:
Connect the interrupt, you'll want to use it.

dirksavage88, 2018-03-25:
Okay, int can be connected to any digital pin the atmega i'm assuming?

closed 2022-10-23 by dirksavage88

GitMoDu, 2022-10-25:
D2 and D3 have a dedicated INTERRUPT so that will have the best performance and simpler use. But any digital pin should work with a PIN_INTERRUPT, provided you have no other pin interrupts or logic to figure out the source.