kriswiner/MPU9250 #252: May have blown my 9250

mjs513, 2018-03-28:
Hi Kris,
Been using one of your 9250 boards extensively for testing a whole range of things :) gyro temperature compensation, a variety of Kalman Filters, SPI_MSTransfer :) etc.  In all cases I have been using the interrupt pin to know when the data is ready.  Never been a problem with long runs - been doing it for awhile - I have been using Brian Taylors 9250 library with changes to use i2c_t3.  That's the background.

Now all of a sudden it seems to stop working at different times (no raw data).  Really first noticed it when the temperature went over 30deg C or so.  Guess my question is, is there a way for me to test if I blew the board or is this expected?  Any suggestions would be helpful.

Thanks
Mike

kriswiner, 2018-03-28:
The MPU9250 is robust up to 85 C so unlikely you could damage the chip. The
solder joints should also be good but if it is one of my boards, I use
low-temperature solder paste (Sn/Bi) and this can oxidize with time. So if
you have been using the board a lot well. maybe the solder joint for the
interrupt just quit, I don;t know. But this is not normal. So i would
conclude the board is busted.

mjs513, 2018-03-28:
Been doing some testing since I posted, if I take the interrupt out of the equation it seems to work.  But I have to let it run for awhile to see if its just the pin or something else.  Hopefully its just a connection issue on my end.  I have been pulling the 9250 out to put on different proto's, but we will see.  I usually am careful when I do that but you never know.  If it is a problem I will be ordering a couple more :) probably will anyway :)

closed 2018-03-28 by mjs513

kriswiner, 2018-03-28:
Should be able to get data without the interrupt.

mjs513, 2018-03-28:
Yeah.  Its the interrupt.  Changed pins and changed jumpers. Same effect.

Like you said I still can get the data without it.  Still going to order another one - hope you have them in stock :)

Thanks
Mike

mjs513, 2018-03-28:
Thanks.  Site said you had 6 left in stock :) so I just ordered 2 more.

What can I tell you - this way I don't have to keep moving them from board to board.

kriswiner, 2018-03-28:
Thanks! I appreciate the business...

mjs513, 2018-03-28:
Any time - love your boards.  You check out what's going on with the uNAVAhrs thread and the linking to GPS.  Then there tonton81's spi transfer library crazy stuff..

kriswiner, 2018-03-28:
" the uNAVAhrs thread and the linking to GPS. Then there tonton81's spi
transfer library crazy stuff.. "

I have no idea what you are talking about. How about a link?

mjs513, 2018-03-28:
Hi Kris
Sorry, I crashed and burned last night so didn't see this until now.  defragster and I are using your MPU-9250 board for all our testing :) Here are the links:

- https://forum.pjrc.com/threads/48450-uNav-AHRS : 4, 7, 15 state Kalman Filters (GPS integration)
- https://forum.pjrc.com/threads/48856-uNav-INS : Kalman filter with GPS integration
- https://forum.pjrc.com/threads/46058-A-UBlox-GPS-Module-Primer-for-beginners : revamp of Ublox library, Chris O. done a lot of work to incorporate setting changes from sketch
- https://forum.pjrc.com/threads/50008-Project-SPI_MSTransfer : tonton81 fast data transfer between teensies, lots and lots of data

Oh, by the way, here is one for you.  I hooked up a completely different 9250 breakout board that I had and same thing happened.  I had the I2C bus at 400K for the tests.  I changed the I2C bus speed back to 100K and it worked.  I re-hooked up your board and its working fine. 

Mike

kriswiner, 2018-03-28:
>    Module-Primer-for-beginners : revamp of Ublox library, Chris O. done a

If you are using the Teensy and i2c_t3.h wire library there could have been
a library-caused glitch somewhere. I had trouble a while back using a
BNO055 that was traced to a bad timer call in the library. In that case, it
worked at some I2C clock speeds but not others. Wasn't limited to the
interrupt though....it was an I2C problem.

mjs513, 2018-03-28:
Thanks for the info Kris.  Good to know.  I have been using the new IDE (1.42b3), I am going to test with the old version just in case as well.