kriswiner/MPU9250 #253: Unable to access accelerometer and gyro data

LuckierBread, 2018-03-28:
Hi Kris, 

I am using the MPU9250 in a custom PCB. with an Arduino Nano as the controller.

VCC --> 3.3 V
GND --> GND
SCL --> A5
SDA --> A4
ADO --> GND

I am unable to read any part of the accelerometer or gyro, all reads come back as 0. 
The Who_Am_I register also returns 0.

I am using a logic level converter to change the 5V Arduino lines to 3.3V for the MPU9250,
and have made sure that the data sent from the Arduino is making it to the SDA and SCLK pins with an oscilloscope. 
When data is requested the line gets held low for as long as it would take to send the requested data.
I have 5K pull up resistors on each of the two lines.

I have confirmed that all my connections are solid.
used an i2c scanner to confirm that a device exists at 0x68.
and successfully retrieved data from the magnetometer.

To eliminate possible code errors I have tried several recommended codes including MPU9250BasicAHRS.ino 
I am currently using test code recommended by kriswiner in a simmaler issue.
code shown here.

`#include <Wire.h>
#include <TimerOne.h>

#define    MPU9250_ADDRESS            0x68
#define    MAG_ADDRESS                0x0C

#define    GYRO_FULL_SCALE_250_DPS    0x00  
#define    GYRO_FULL_SCALE_500_DPS    0x08
#define    GYRO_FULL_SCALE_1000_DPS   0x10
#define    GYRO_FULL_SCALE_2000_DPS   0x18

#define    ACC_FULL_SCALE_2_G        0x00  
#define    ACC_FULL_SCALE_4_G        0x08
#define    ACC_FULL_SCALE_8_G        0x10
#define    ACC_FULL_SCALE_16_G       0x18

// This function read Nbytes bytes from I2C device at address Address. 
// Put read bytes starting at register Register in the Data array. 
void I2Cread(uint8_t Address, uint8_t Register, uint8_t Nbytes, uint8_t* Data)
{
  // Set register address
  Wire.beginTransmission(Address);
  Wire.write(Register);
  Wire.endTransmission();
  
  // Read Nbytes
  Wire.requestFrom(Address, Nbytes); 
  uint8_t index=0;
  while (Wire.available())
    Data[index++]=Wire.read();
}

// Write a byte (Data) in device (Address) at register (Register)
void I2CwriteByte(uint8_t Address, uint8_t Register, uint8_t Data)
{
  // Set register address
  Wire.beginTransmission(Address);
  Wire.write(Register);
  Wire.write(Data);
  Wire.endTransmission();
}

// Initial time
long int ti;
volatile bool intFlag=false;

// Initializations
void setup()
{
  // Arduino initializations
  Wire.begin();
  Serial.begin(115200);
  // Set accelerometers low pass filter at 5Hz
  I2CwriteByte(MPU9250_ADDRESS,29,0x06);
  // Set gyroscope low pass filter at 5Hz
  I2CwriteByte(MPU9250_ADDRESS,26,0x06);
 
  // Configure gyroscope range
  I2CwriteByte(MPU9250_ADDRESS,27,GYRO_FULL_SCALE_1000_DPS);
  // Configure accelerometers range
  I2CwriteByte(MPU9250_ADDRESS,28,ACC_FULL_SCALE_4_G);
  // Set by pass mode for the magnetometers
  I2CwriteByte(MPU9250_ADDRESS,0x37,0x02);
  
  // Request continuous magnetometer measurements in 16 bits
  I2CwriteByte(MAG_ADDRESS,0x0A,0x16);
  
   pinMode(13, OUTPUT);
  Timer1.initialize(10000);         // initialize timer1, and set a 1/2 second period
  Timer1.attachInterrupt(callback);  // attaches callback() as a timer overflow interrupt
  
  
  // Store initial time
  ti=millis();
}

// Counter
long int cpt=0;

void callback()
{ 
  intFlag=true;
  digitalWrite(13, digitalRead(13) ^ 1);
}

// Main loop, read and display data
void loop()
{
  while (!intFlag);
  intFlag=false;
  
  // Display Time

  Serial.print (millis()-ti,DEC);
  Serial.print ("\t");

  
  // _______________
  // ::: Counter :::
  
  // Display data counter
//  Serial.print (cpt++,DEC);
//  Serial.print ("\t");
 
  // ____________________________________
  // :::  accelerometer and gyroscope ::: 

  // Read accelerometer and gyroscope
  uint8_t Buf[14];
  I2Cread(MPU9250_ADDRESS,0x3B,14,Buf);
  
  // Create 16 bits values from 8 bits data
  
  // Accelerometer
  int16_t ax=-(Buf[0]<<8 | Buf[1]);
  int16_t ay=-(Buf[2]<<8 | Buf[3]);
  int16_t az=Buf[4]<<8 | Buf[5];

  // Gyroscope
  int16_t gx=-(Buf[8]<<8 | Buf[9]);
  int16_t gy=-(Buf[10]<<8 | Buf[11]);
  int16_t gz=Buf[12]<<8 | Buf[13];
  
    // Display values
  
  // Accelerometer
  Serial.print (ax,DEC); 
  Serial.print ("\t");
  Serial.print (ay,DEC);
  Serial.print ("\t");
  Serial.print (az,DEC);  
  Serial.print ("\t");
  
  // Gyroscope
  Serial.print (gx,DEC); 
  Serial.print ("\t");
  Serial.print (gy,DEC);
  Serial.print ("\t");
  Serial.print (gz,DEC);  
  Serial.print ("\t");

  
  // _____________________
  // :::  Magnetometer ::: 

  
  // Read register Status 1 and wait for the DRDY: Data Ready
  
  uint8_t ST1;
  do
  {
    I2Cread(MAG_ADDRESS,0x02,1,&ST1);
  }
  while (!(ST1&0x01));

  // Read magnetometer data  
  uint8_t Mag[7];  
  I2Cread(MAG_ADDRESS,0x03,7,Mag);
  

  // Create 16 bits values from 8 bits data
  
  // Magnetometer
  int16_t mx=-(Mag[3]<<8 | Mag[2]);
  int16_t my=-(Mag[1]<<8 | Mag[0]);
  int16_t mz=-(Mag[5]<<8 | Mag[4]);
  
  
  // Magnetometer
  Serial.print (mx+200,DEC); 
  Serial.print ("\t");
  Serial.print (my-70,DEC);
  Serial.print ("\t");
  Serial.print (mz-700,DEC);  
  Serial.print ("\t");
  
  
  
  // End of line
  Serial.println("");
//  delay(100);    
}`

kriswiner, 2018-03-28:
void readBytes(uint8_t address, uint8_t subAddress, uint8_t count, uint8_t
* dest)
{
Wire.beginTransmission(address); // Initialize the Tx buffer
Wire.write(subAddress); // Put slave register address in Tx buffer
Wire.endTransmission*(false*); // Send the Tx buffer, but send a restart to
keep connection alive
uint8_t i = 0;
Wire.requestFrom(address, count); // Read bytes from slave register address
while (Wire.available()) {
dest[i++] = Wire.read(); } // Put read results in the Rx buffer
}

LuckierBread, 2018-03-28:
Thank you for the timely response. I have made the recommended change, However, I still receive constant  0s from the MPU9250.

kriswiner, 2018-03-28:
Try rea

kriswiner, 2018-03-28:
Doesn't make sense you can read the mag butnot accel.

LuckierBread, 2018-03-28:
Is it possible that the ic is fried? normally I wouldn't expect an ic to report back anything if it was fried.

kriswiner, 2018-03-28:
Always possible

LuckierBread, 2018-03-28:
Stranger still is that this code I wrote as an earlier test seems to return random values ever once and a while.
Column order accelerometer x,y,z gyro x,y,z.
[image]

`#include<Wire.h>
const int MPU_addr=0x68;  // I2C address of the MPU-6050
int Tmp, ax,ay,az,gx,gy,gz;

void setup(){
  Wire.begin();
  Wire.setClock(400000);
  Wire.beginTransmission(0x68);
  Wire.write(0x6B);  // PWR_MGMT_1 register
  Wire.write(0x00);     // set to zero (wakes up the MPU-6050)
  Wire.write(0x00);     //set PWR_MGMT_2 register to full operation
  Wire.endTransmission(true);
  Serial.begin(9600);
}
void loop(){
  Wire.beginTransmission(4);
  Wire.write(0x3B);  // starting with register 0x3B (ACCEL_XOUT_H)
  Wire.endTransmission(false);
  Wire.requestFrom(MPU_addr,14,true);  // request a total of 14 registers
  ax=Wire.read()<<8|Wire.read();  // 0x3B (ACCEL_XOUT_H) & 0x3C (ACCEL_XOUT_L)     
  ay=Wire.read()<<8|Wire.read();  // 0x3D (ACCEL_YOUT_H) & 0x3E (ACCEL_YOUT_L)
  az=Wire.read()<<8|Wire.read();  // 0x3F (ACCEL_ZOUT_H) & 0x40 (ACCEL_ZOUT_L)
  Tmp=Wire.read()<<8|Wire.read();  // 0x41 (TEMP_OUT_H) & 0x42 (TEMP_OUT_L)
  gx=Wire.read()<<8|Wire.read();  // 0x43 (GYRO_XOUT_H) & 0x44 (GYRO_XOUT_L)
  gy=Wire.read()<<8|Wire.read();  // 0x45 (GYRO_YOUT_H) & 0x46 (GYRO_YOUT_L)
  gz=Wire.read()<<8|Wire.read();  // 0x47 (GYRO_ZOUT_H) & 0x48 (GYRO_ZOUT_L)

  // Accelerometer
  Serial.print (ax,DEC); 
  Serial.print ("\t");
  Serial.print (ay,DEC);
  Serial.print ("\t");
  Serial.print (az,DEC);  
  Serial.print ("\t");
  
  // Gyroscope
  Serial.print (gx,DEC); 
  Serial.print ("\t");
  Serial.print (gy,DEC);
  Serial.print ("\t");
  Serial.print (gz,DEC);  
  Serial.println();
 delay(100);
}`

kriswiner, 2018-03-28:
Wire.beginTransmission(4);

Why 4 here?

You need to first write the address Wire.write(0x68);

This is prett sloppy code writing.

Why not try something like this
<https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_BasicAHRS2_ESP8266.ino>
example?

Or at least take the I2C read/write functions from here.

LuckierBread, 2018-04-05:
Hi Kris, I'm back. 
I acknowledge that the last post's code was messy. I'm sorry for that.

Using the code that I displayed in my original post with the changes you recommended. 
I have ruled out that the error is the fault of the mpu9250 IC. I ordered a replacement and it is having the same problem.  I am using a breakout board from SparkFun found here: [https://www.sparkfun.com/products/13762](url)
The breakout board is connected to a custom PCB the net that it is connected to can be seen here:
[image]
Note that there is a BQ32002 Real-time clock also on the i2c line on the 3.3V side of the logic converter. however, it's address should be 0xC0 so that shouldn't really affect it.
The Bi-Directional Logic Converter circuit is here:
[image]
I am having a difficult time figuring out if this is a hardware issue or a software issue. 
Please let me know if you see a glaring issue that I am missing here.

kriswiner, 2018-04-05:
Are you connecting 3V3 to VDDIO?

LuckierBread, 2018-04-05:
yes, the break out board has them bound together.

kriswiner, 2018-04-05:
This is a custom pcb, but using Sparkfun's breakout?

LuckierBread, 2018-04-05:
yes, I am using SparkFun's breakout because I did not feel confident soldering the very small pins on the MPU9250 with my soldering iron.

LuckierBread, 2018-04-05:
Wow. That is embarrassing. I have no idea how I missed that. Thank you for spotting that.
I don't understand how I was able to read the magnetometer and not the other stuff though that.

kriswiner, 2018-04-05:
Why are you using a 5 V AVR? Why not a 3V3 STM32l4?

LuckierBread, 2018-04-05:
The honest answer is ignorance. My course doesn't go in depth for picking processers. For a project that is worth 80% of my grade, I didn't want to risk using a processor that i was unfamiliar with as the core of my project.

kriswiner, 2018-04-05:
This might be worth the trouble:

https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/?pt=ac_prod_search

This has a 1 ppm RTC built in, and is 3V3, and will allow fusion at a 20
kHz rate. Arduino core, easy to use, lot's of examples, etc.

kriswiner, 2018-04-05:
> I don't really know what you mean by fusion. It looks like it has some
> very nice specs. If it accepts the same code format as the nano then I
> might have to swap to that for the V2 of this project.
> Also Thank you very much for your time. It ended up having nothing to do
> with your code but you helped anyway. You have both my gratitude and
> respect.

You don't intend to use the mPU9250 data to estimate absolute orientation?

LuckierBread, 2018-04-05:
oh, that's what you meant. I definitely am trying to get the absolute orientation. I am building a self-balancing one-wheeled skateboard. the wheel is powered and will be controlled like a Segway. just lean in the direction that you want to go. I picked the MPU9250 because of the built-in DMP.

kriswiner, 2018-04-05:
DMP cannot do 9 DoF fusion, only roll and pitch. Maybe this is good enough.
But you get the fusion out at 100 Hz, that;'s it.

Good luck using it...

closed 2018-04-05 by LuckierBread