kriswiner/MPU9250 #254: BMP Conversion Shifting Issue

wansco, 2018-03-30:
I'm not sure if this is an issue (if its corrected elsewhere), but there appears to be a shifting bug when reading the temperature and pressure from the BMP280 in MPU9250_BMP280_BasicAHRS_t3.ino

`return (int32_t) (((int32_t) rawData[0] << 16 | (int32_t) rawData[1] << 8 | rawData[2]) >> 4);`
which results in zeros for bits 4:7

The layout in the bmp280 is kindof screwy, and it looks like it should be something like this:

`return (int32_t) (((int32_t) rawData[0] << 12 | (int32_t) rawData[1] << 4 | rawData[2]) >> 4);`

kriswiner, 2018-03-30:
No, the idea is to cast the three bytes into a signed 20-bit integer.

This:

return (int32_t) (((int32_t) rawData[0] << 12 | (int32_t) rawData[1] << 4 |
rawData[2]) >> 4);

is not correct. For one thing, then rawData[1] and rawDate[2] collide...

wansco, 2018-03-30:
oooops, nevermind, now I see what you're doing (filling from the left, then shifting everything down by 4). I missed that 2nd to last parentheses. I should have said this, but on closer inspection, either should work

`return (int32_t) ((int32_t) rawData[0] << 12 | (int32_t) rawData[1] << 4 | rawData[2] >> 4);`

closed 2018-03-30 by wansco

kriswiner, 2018-03-30:
The above won't work...mis-placed parathensis