kriswiner/MPU9250 #255: pitch angle is not accurate.

pradeep14581, 2018-03-31:
hi thanks for such a wonderful library. I am using CC2650 launchpad (cortex arm m3) of Texas Instruments   and i am using your library for reading the data from mpu9250. I have calibrated the mpu9250 as  givrn in the following link 
https://github.com/kriswiner/MPU9250/issues/120
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration
https://github.com/kriswiner/MPU9250/issues/126

and when mpu9250 is at rest and flat, i am getting accelerometer value as
0.001, -0.07, 1.01 
0.002, -0.10,1.03 etc
gyrometer value as 
0.10, -0.32, 0.9 
0.5, -0.41, 0.6 etc
and magnetometer value  
 209.9145 323.8681 371.8486
 215.9120 335.8632 371.8486
 206.9157 323.8681 353.8559
 215.9120 332.8644 359.8534 etc. 
i do not know how to verify the magnetometer value whether it is correct or not. if is there any way please tell me.
after calibration i am getting gyro bias 1.0381 -4.2290 1.8396 accelerometer bias  -0.1036 0.0220 -0.0525 and magnetometer 
  bias                      scale 
54    391  112        1.19   0.94  0.90
51   384    150        1.5   0.9     0.86 
35    457    152         1.3  0.88   0.82 etc
when i calculated the roll, pitch and angle i am getting roll and yaw angle fine but getting error in pitch angle. i am getting 75-80 degree only when i rotate around y-axis. when i rotate mpu9250 around x-axis only roll angle changes and nominal change in pitch and yaw angle(around 20-30 degree). and same for the yaw angle also. But when i rotate around y-axis pitch angle is changing along with yaw also changing drastically but nominal change in roll angle. i performed these test by rotating mpu9250 90 and 180 degree 
thank you.

pradeep14581, 2018-03-31:
no i am not getting negative value i am getting following values

 mag 445.5334 152.9786 223.9747
 mag 460.2188 144.6102 231.6708
 mag 477.0021 147.9576 222.4355
 mag 449.7292 139.5891 217.8179
 mag 468.6105 147.9576 213.2002
 mag 472.8063 157.9998 216.2787
 mag 462.3167 152.9786 223.9747
 mag 460.2188 151.3050 216.2787
 mag 483.2958 149.6313 227.0532
 mag 472.8063 147.9576 222.4355
 mag 466.5125 132.8943 220.8963
 mag 449.7292 149.6313 214.7394
 mag 453.9250 146.2839 223.9747

kriswiner, 2018-03-31:
What is the z-axis value when flat on a table?  And then what is the
z-value of the mag when the board is turned upside down?

pradeep14581, 2018-03-31:
values when flat on table
mag 315.4626 -275.4868 562.6025
 mag 315.4626 -285.5290 559.5241
 mag 330.1480 -253.7288 536.4358
 mag 319.6584 -272.1394 559.5241
 mag 311.2668 -275.4868 556.4456
 mag 323.8543 -268.7920 544.1319
 mag 323.8543 -262.0972 556.4456
 mag 323.8543 -265.4447 556.4456
 mag 319.6584 -272.1394 553.3672
 mag 321.7564 -273.8131 554.9064
 mag 323.8543 -252.0551 547.2103
 mag 334.3439 -270.4657 557.9849
 mag 315.4626 -258.7499 550.2888
 mag 321.7564 -267.1183 557.9849

these are the values when board is  turned upside down
 mag 309.1689 157.9998 -251.6432
 mag 288.1897 131.2207 -267.0354
 mag 296.5814 134.5680 -273.1922
 mag 283.9939 151.3050 -267.0354
 mag 286.0918 132.8943 -271.6530
 mag 283.9939 134.5680 -273.1922
 mag 281.8960 139.5891 -277.8099
 mag 294.4835 129.5470 -280.8883
 mag 283.9939 141.2628 -263.9570
 mag 273.5043 142.9365 -287.0452
 mag 298.6793 139.5891 -271.6530
 mag 279.7981 161.3471 -267.0354
 mag 265.1127 132.8943 -274.7315
 mag 277.7001 142.9365 -274.7315
 mag 298.6793 149.6313 -265.4962
 mag 281.8960 166.3682 -271.6530

kriswiner, 2018-03-31:
Not calibrated

pradeep14581, 2018-03-31:
this is calibrated

pradeep14581, 2018-03-31:
with mag bias 35 457 152, 1.3 0.88 0.82

kriswiner, 2018-03-31:
Did you move the mag all around in every 3D direction to get these numbers?
You mag is not currently calibrated, based on the numbers you are showing
me...

pradeep14581, 2018-03-31:
yes i move magnetometer just like a small spherical motion and i calibrated it many times but every time i get little different value as following

54 391 112 , 1.19 0.94 0.90
51 384 150, 1.5 0.9 0.86
35 457 152, 1.3 0.88 0.82 etc

pradeep14581, 2018-03-31:
ok i will try to calibrate it again

pradeep14581, 2018-03-31:
by the way is there any way by which which we can verify whether the magnetometer value is correct or not

kriswiner, 2018-03-31:
And then you are subtracting these values from your data?

In this case, the Mz test should show the same but opposite sign when the
board is face up versus face down.

pradeep14581, 2018-03-31:
ok thanks for the reply. i am checking it right now

pradeep14581, 2018-03-31:
right now i got this data after calibration when mpu9250 flat up and flat down. but the pitch angle is still not correct. 

mag 136.0249 171.1975 -443.3781
 mag 160.4396 175.0018 -430.3853
 mag 143.0005 186.4150 -433.6335
 mag 149.9762 178.8062 -430.3853
 mag 151.7201 157.8821 -432.0094
 mag 158.6958 169.2953 -425.5130
 mag 156.9518 167.3930 -440.1299
 mag 146.4884 163.5887 -433.6335
 mag 143.0005 178.8062 -453.1227
 mag 144.7445 173.0997 -415.7684
 mag 148.2323 165.4909 -419.0166
 mag 141.2566 169.2953 -445.0022
 mag 143.0005 175.0018 -446.6262
 mag 148.2323 165.4909 -425.5130
 mag 153.4640 175.0018 -430.3853
 mag 153.4640 175.0018 -433.6335
 mag 143.0005 175.0018 -430.3853
 mag 146.4884 171.1975 -436.8817
 mag 144.7445 169.2953 -435.2576
 mag 148.2323 180.7084 -445.0022
 mag 153.4640 171.1975 -475.8600
 mag 137.7688 214.9480 -435.2576
 mag 76.7320 403.2651 -238.7420
 mag 97.6589 323.3730 183.5228
 mag 153.4640 148.3711 397.9034
 mag 111.6102 45.6526 449.8744
 mag 101.1467 11.4131 423.8889
 mag 81.9637 -1.9021 441.7540
 mag 95.9150 9.5109 435.2576
 mag 85.4515 1.9021 451.4985
 mag 87.1954 0.0000 430.3853
 mag 90.6833 7.6087 407.6480
 mag 104.6345 -11.4131 397.9034
 mag 85.4515 -17.1197 402.7756
 mag 97.6589 -3.8044 410.8961
 mag 87.1954 0.0000 407.6480
 mag 87.1954 -11.4131 423.8889
 mag 95.9150 -5.7065 412.5202
 mag 87.1954 0.0000 417.3926
 mag 73.2442 -7.6087 404.3998
 mag 85.4515 -1.9021 428.7612
 mag 97.6589 -7.6087 440.1299

kriswiner, 2018-03-31:
Much better

pradeep14581, 2018-04-01:
I am not getting pitch angle correct even after calibrating many times. What would be the problem?

kriswiner, 2018-04-01:
Are you feeding the fusion algorithm with sensor data in the correct NED or
ENU order?

pradeep14581, 2018-04-01:
MadgwickQuaternionUpdate(-ax, ay, az, gx*3.14/180.0f, -gy*3.14/180.0f, -gz*3.14/180.0f,  my,  -mx, -mz,deltat);

right now i am providing this NED order and i think above is correct NED order but for testing purpose i have used following also when above one was not giving me the correct result. 

 //MadgwickQuaternionUpdate(ax, ay, az, gx*3.14/180.0f, gy*3.14/180.0f, gz*3.14/180.0f,  my,  mx, mz,deltat);
 
 // MadgwickQuaternionUpdate(ax, ay, az, gx*3.14/180.0f, -gy*3.14/180.0f, -gz*3.14/180.0f,  my,  -mx, mz,deltat);
// MadgwickQuaternionUpdate(ay, ax, -az, gy*3.14/180.0f, gx*3.14/180.0f, -gz*3.14/180.0f,  mx,  my, mz,deltat);

kriswiner, 2018-04-01:
Compass +Mz is down, above cannot be NED

pradeep14581, 2018-04-02:
ok so what would be the correct NED.

kriswiner, 2018-04-02:
Exactly...consult the product spec to determine.

pradeep14581, 2018-04-02:
[image]

according to the data sheet all the possible NED would be as following.

(ay, ax,-az,  gy, gx, -gz, mx, my, mz)

(ax, -ay, -az,  gx, -gy, -gz, my, -mx, mz)

(-ay , -ax, -az,  -gy, -gx, -gz, -mx, -my, mz)

(-ax,  ay, -az, -gx, gy, -gz, -my, mx, mz)
 
(az, ay, ax, gz, gy, gx, -mz, mx, my)

i have tried all the above combination but still the problem is with pitch angle only.

kriswiner, 2018-04-02:
This: (ay, ax,-az, gy, gx, -gz, mx, my, mz)  and well calibrated sensors
should work. If not, youmight have a defective sensor, but this seems
unlikely.

What are you using for an MCU? It should be able to run the fusion filter
at 1 kHz or so to get accurate orintation estimation.

I would use 200 Hz gyro and accel rate, 100 Hz mag rate, and a fast enough
MCU to get > 1 kHz fusion rate. I have done this dozens if noit hundreds of
times and always gotten a good result. Not sure  what could be your problem
if not these things.

pradeep14581, 2018-04-02:
I am using CC2650 of Texas instruments having 48 MHz.
for gyro and accelerometer 200 HZ
and for magnetometer 100 HZ

pradeep14581, 2018-04-03:
HI, the problem is still not resolved i tried it with arduino also but not getting the correct pitch angle.

pradeep14581, 2018-04-03:
Ok understood the pitch will range from -90 degree to 90 degree because of sin(), but why yaw is changing in case of pitch rotation. Another thing  i am not getting correct linear acceleration, when it is flat and down and motionless its giving me -0.21, 0.3, -2.34 . this is really unexpected

kriswiner, 2018-04-03:
If this is in milli gs this looks perfectly normal to me.

pradeep14581, 2018-04-05:
No its in terms of g not in milli g

kriswiner, 2018-04-05:
Maybe you are subtracting 1 g from the z-axis bias instead of adding in the
calibration? This will depend ong the orientation of the sensor when flat
and motionless. The sketch is for the sensor pointing up, on top of the
board. I suspect you have your sensor on the bottom.

pradeep14581, 2018-04-05:
Yes my sensor is on the bottom and i am using  following lines to get the linear acceleration
 
float lin_ax = ax + a31;

    float lin_ay = ay + a32;

    float lin_az = az - a33;//az+a33;

kriswiner, 2018-04-05:
Likely the accel calibration routine needs to be adjusted. 1 g is
subtracted from the z-axis biard, you need to add it.

pradeep14581, 2018-04-05:
ok thanks, i'll add it

pradeep14581, 2018-04-05:
ok i got it actually when i am running at full range  gres  2000dps,  ares  16G  then i am getting wrong values of pitch and linear acceleration but if i am running at range  250 dps and 2G then its working fine. By the way did you ever test the mpu9250 on full range.
Thank you

kriswiner, 2018-04-05:
Looks like your scaling is off then.