kriswiner/MPU9250 #257: Delta Angles are not correct. What is most likely the problem?

Mcdane, 2018-04-03:
Hi Kris

Thanks for your work. I have integrated some of your code into mine. The whole code seems to work quite well with the mpu9250. However, there is one problem: let's suppose the mpu9250 is flat on the table when

- I turn it 90 degrees to the right, it will make a **delta of 86 degrees**

- I turn it again 90 degrees to the right (180 in total so far), it will make a **delta of 91 degrees**

- I turn it again 45 degrees to the right (225 in total so far), it will make a **delta of 53 degrees**

- I turn it again 45 degrees to the right (270 in total so far), it will make a **delta of 40 degrees**

-  I turn it again 45 degrees to the right (315 in total so far), it will make a **delta of 47 degrees**

-  I turn it again 45 degrees to the right (360 in total so far), it will make a **delta of 43 degrees**

Now when you sum it up for this specific example, you get: **86+91+53+40+47+43 = 360**.

But ideally it should be: 90+90+45+45+45+45=360

Although I calibrated the magnetometer using an ellipsoid fit (and data looks like a sphere), my two guesses are either the mag. has to be calibrated even better or my inputs for the MahonyFilter is not correct. But please check some details below:

===========================================

**The Details**

Let me give you some specifications that I have:

1. I am using an **Arduino DUE**.

2. I am reading the **Gyro** (DLPF: 3dB-Bandwidth=>41Hz) and the **Accelerometer** (DLPF: 3dB-Bandwidth=>44.8Hz) at a **sampling frequency  fs = 1000Hz** (SAMPLE RATE DEVIDER Register Value==> 0x00).

3. I am running the **AK8963** in "16 bit - continious mode 2". I guess this is a sampling freq of 100Hz. right?

4. I am using the Mahony Filter with these **Gains**

#define Kp 2.0f * 5.0f * 10.0f
#define Ki 0.0f

I had the feeling that these values provided the best results (fast responsive).

5. My **Mahony-Inputs** are as such:

MahonyQuaternionUpdate(ax, -ay, -az, gx*(M_PI)/180.0f, -gy*(M_PI)/180.0f, -gz*(M_PI)/180.0f, mag_y, -mag_x, mag_z, deltaT);

When I calibrated gyro and acc, all values were "set to zero" and accelerometer's z-Axis was pointing to the sky and showing "-1g" (negative 1g).

Is this all correct? 

6. **Tilt-Compensation** is more or less ok: within a couple of degrees ranging from **1...5 degrees** (depending on the heading angle).

7. I calibrated the magnetometer using an **ellipsoid fit**. These are the three 2d plots (x-y, y-z, x-z):

[image]

[image]

[image]

kriswiner, 2018-04-03:
>    1. My *Mahony-Inputs* are as such:

This all seems right to me. Try calculating the standard deviation of your
90-degree turning test, looks like it will be ~5 degrees or so, which is
about what one can expect for these simple fusion filters.

You might do better by increasing the fusion rate, likely with a faster
processor, but typical is ~4 degree rms heading accuracy with the Madgwick
or Mahony algorithms and properly calibrated sensors.

This
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/>
device can routinely obtain two degree rms heading accuracy.

One can do better still (~1 degree rms heading accuracy) using more
sophisticated magnetometer calibration methods and proprietary fusion
filtering.

Mcdane, 2018-04-03:
Thanks. And you also think that this is correct:

My Mahony-Inputs are as such:
MahonyQuaternionUpdate(ax, -ay, -az, gx*(M_PI)/180.0f, -gy*(M_PI)/180.0f, -gz*(M_PI)/180.0f, mag_y, -mag_x, mag_z, deltaT);

When I calibrated gyro and acc, all values were "set to zero" and accelerometer's z-Axis was pointing to the sky and showing "-1g" (negative 1g) ?

Because from other posts I also read:

(ax, ay, az, gx, gy, gz, my, mx, -mz)

I really got confused by NED and Acc. Calibration (inital gravity orientation) and how to correctly apply it.

kriswiner, 2018-04-03:
OK, maybe time for a little tutorial. See the attached from the MPU9250
Product Specification.

These filters expect a consistent ordering of the data, NED is usually what
I use, but ENU will also work. The key is consistency.

Lets say I want NED and I choose North to be defined as the Ax axis.

Then -Ay lies points East, and -Az is Down. So the accel terms should be
fed into the fusion filter as (Ax, -Ay, -Az, ...)

Hey, that's what you have! Good job. Gyro should be the same, since accel
and gyro axes are aligned.

Now for the mag. If Ax is aligned with North, then My is aligned with
North. -Mx is aligned with East, and Mz is pointing Down.

Notice that you can choose a different convention, like ENU, or choose a
different axis pointing North, like -Ay, and simply repeat the procedure.
As long as the order is consistent with your convention, you should get
good results.

One thing I do not understand is your gravity vector. If the device package
is on top of the board, by convention Earth's gravity should show +1 g,
even though gravity acts in the downward direction. For this reason, I
usually negate all of the accel (just the accel) values.

Mcdane, 2018-04-03:
**The Gravity Vector**

When the sensor is flat on the table and z-axis point to the sky, there is the gravity force pointing in the other direction. Hence, I am telling the sensor in this moment where "Down" is by saying "there is a force pointing against the z-axis (0, 0, -1).

[image]

kriswiner, 2018-04-03:
Nope...

Mcdane, 2018-04-03:
So according to what you wrote previously it should initially be set to (0, 0, 0)?

kriswiner, 2018-04-03:
0, 0, 1 is the convention.

Mcdane, 2018-04-03:
I get wrong tilt-compensation with (0, 0, 1) & NED. But "correct" (within a couple of degrees as described above) with (0, 0, -1) & NED. :-/

kriswiner, 2018-04-03:
We then use 0, 0, -1. I usually negate the accel sensor values, maybe this
is the same thing as whatever yu are doing.

kriswiner, 2018-04-04:
> Hey thanks a lot for your effort.
> Btw can you give me some hints regarding

Not yet...

Mcdane, 2018-04-05:
ok thanks :-)