kriswiner/MPU9250 #258: data type of magnetometer ASAX, ASAY and ASAZ

jimnlx, 2018-04-03:
I see in the MPU-9250 code that ASAX, ASAY and ASAZ are treated as 8-bit unsigned values. Was this data type obtained from the manufacturer? I couldn't find confirmation in documentation about whether this 8-bit value was signed or unsigned. If it's unsigned, the range of adjustment is 0.5 to 1.5. If it's signed, the range of adjustment is 0.0 to 1.0. A range of 0.5 to 1.5 seems more practical.

Annoyingly, the ASAX, ASAY and ASAZ reported by my magnetometer result in passing the magnetometer self test in both cases (8-bit unsigned and 8-bit signed)!

Thanks,
Jim

kriswiner, 2018-04-03:
The mag is the AK8963C, take a look at the data sheet.

The ASAs are factory calibrations of the axes response, not self test
parameters.

jimnlx, 2018-04-03:
Hi Kris,

Thanks. I know that the mag is the AK8963C. I also know the ASAs are for the axes response. To get an accurate self test, the ASAs should be applied, shouldn't they?

My question was about whether the was documentation somewhere that clearly called out the ASAs as 8-bit unsigned as opposed to 8-bit signed.

kriswiner, 2018-04-03:
single bytes are not signed.

jimnlx, 2018-04-03:
Interesting!

Thanks,
Jim