kriswiner/MPU9250 #259: Pitch and Roll seems fine but Yaw is funny ;)

TByte007, 2018-04-07:
lets say I get those by default 
Yaw, Pitch, Roll: -86.63, 4.21, 179.56
If I turn the sensor around I get close enough values for the Pitch and Roll but this is what I get when the sensor is at 90 degrees horizontally to the position above:
Yaw, Pitch, Roll: -102.18, 3.22, -179.04

TByte007, 2018-04-07:
It seems like MPU9250BasicAHRS.ino is far from your last version. I added the mag calibration and it now works (with some twitching).  Can u point me to your last working example which incorporates all of your latest changes ?

kriswiner, 2018-04-07:
None of these sketches is intended to be a black box. You are going to have
to adjust the parameters for your particular MCU and use case.

This
<https://github.com/kriswiner/Ladybug/blob/master/MPU9250_MS5637_BasicAHRS2_Ladybug.ino>
is a recent sketch, and the library is here
<https://github.com/kriswiner/MPU9250/tree/master/MPU9250_BME280_SPIFlash_Ladybug>