kriswiner/MPU9250 #260: initAK8963() the parameter _Mmode is shadowing the object's variable ?

TByte007, 2018-04-07:
`void MPU9250::initAK8963(uint8_t _Mscale, uint8_t _Mmode, float * _magCalibration)`
this is the original non working definition (at least for me). _Mmode is used locally for the function and never set for the object. Then at the calibration routine the object's   _Mmode is zero and of course it never works. 

if I change the init to:
```
void MPU9250::initAK8963(uint8_t _Mscale, uint8_t Mmode, float * _magCalibration)
{
  _Mmode = Mmode;
```
It now appears to work and of course the global (for the object) variable _Mmode is set to 6 (as per your example).

kriswiner, 2018-04-07:
Which sketch?

TByte007, 2018-04-07:
in MPU9250.cpp

kriswiner, 2018-04-07:
Looks like none of those should be globals...

TByte007, 2018-04-07:
_Mmode is defined as "global" for the object MPU9250 in MPU9250.h on line  233 as  uint8_t _Mmode;
And it is used in magcalMPU9250() to set the delay as in "if(_Mmode == 0x02) delay(135);" and the number of samples as in "if(_Mmode == 0x02) sample_count = 128;" . And if u dont set it in initAK8963() or as MPU9250._Mmode = M_100Hz (which will be double the work and why would u do that?) the calibration routine will never work as it will just skip the calibration cycle.

kriswiner, 2018-04-07:
Thanks, been a while since I've used this...

TByte007, 2018-04-07:
Is there a better library that you use then ? :)

kriswiner, 2018-04-07:
You don't like this one?

kriswiner, 2018-04-07:
> I have dozens each for a different combo of sensors and a different MCU.
>
> None for the Nano, though...I stopped using AVRs years ago.

There is this one, I don't use it but maybe you'll like it better...

https://github.com/brianc118/MPU9250

TByte007, 2018-04-07:
I intend to use the sensor(s) with STM32 which will do some of the job (all of the AHRS processing and filtering may be)  and send it to some bigger ARM "machine" with more RAM (like RP or some of the clones) which will do the higher level processing. But it's easier for me to test the lib on Nano as it's hooked to the PC and need only one USB (no STLink) :)

kriswiner, 2018-04-07:
These are what you need then:

https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/

or

https://www.tindie.com/products/TleraCorp/dragonfly-stm32l47696-development-board/

TByte007, 2018-04-07:
Yes those would be perfect for what im trying to do (although I'll try it first with F108 "bluepill" coz I have 3 :) ) . It's just that the shipping to EU doubles the costs :( (and I'm in the eastern part - read "we dont get as much as the western part of EU gets" so cost matters :) )

kriswiner, 2018-04-07:
Shipping does cost what it costs, but you can ship as many items as you'd
like for one shipping charge. Maybe you can get some of your friends
interested to lower the per item shipping cost.

The Dragonfly L496 Vvariant has 1 MByte flash and 320 kByte SRAM. Might be
all you need for your "big-boy" application.

TByte007, 2018-04-07:
I will need lot more RAM coz it will work with camera images and CV (the bot have to recognize where he is in the room and reset the inertial guidance system's error) :) . That means few MB of RAM and relatively heavy libs most of which will require OS and good memory management.
Btw those STM32L4s are really nice (I'm clicking around your store drooling lol) . But the price + shipping will get twice as high as OrangePi with H2+ with 256MB of RAM :) I wonder how the Chinese can manage such a low shipping rates - I get most things for free in like 10 days ...

Btw I found another problem with 
`void MPU9250::initAK8963(uint8_t _Mscale, uint8_t _Mmode, float * _magCalibration)`
now it shadows the objects own "_magCalibration" (same names). I guess those are artifacts from switching to OOP.
I quick fixed it by changing the declaration to 
`void MPU9250::initAK8963(uint8_t _Mscale, uint8_t _Mmode, float * magCalibration)`
and then 
  _magCalibration[0] =  (float)(rawData[0] - 128)/256.0f + 1.0f;   // Return x-axis sensitivity adjustment values, etc.
  _magCalibration[1] =  (float)(rawData[1] - 128)/256.0f + 1.0f;  
  _magCalibration[2] =  (float)(rawData[2] - 128)/256.0f + 1.0f; 

to
```
  magCalibration[0] = _magCalibration[0] =  (float)(rawData[0] - 128)/256.0f + 1.0f;   // Return x-axis sensitivity adjustment values, etc.
  magCalibration[1] = _magCalibration[1] =  (float)(rawData[1] - 128)/256.0f + 1.0f;  
  magCalibration[2] = _magCalibration[2] =  (float)(rawData[2] - 128)/256.0f + 1.0f; 

```
Otherwise the biases wont get calculated as _magCalibration[] will be all zeroes.

kriswiner, 2018-04-07:
Yes, this is odd since I am sure this sketch used to work...

TByte007, 2018-04-10:
I ended up finding out why it worked for you :) . If u preset the values of magBias and magScale (as u did) you end up never using the magcalMPU9250() function so it actually works in that case. The code needs cleaning coz some parts it seems are left inside the objects others are pulled outside and it became kind of a ... you know what :) . Btw using the interrupt and sleep/wake up never worked stable for me. I could see the magnetometer not working fro seconds then just waking up giving few different values then getting to bed again so I removed all the wake up/sleep parts for my case.