kriswiner/MPU9250 #262: Yaw angle is not accurate!

clhne, 2018-04-08:
hello kris,
thanks for such a wonderful library. I am using STM32F103C8T6 and i am using your library for reading the data from mpu9250. 
but,  i find the yaw angle is changed between -180~+180, so what's the problems? 
btw, i have calibrated the magnetometer, but the result don't look right:
[image]
 
thank you.

kriswiner, 2018-04-08:
Mag not calibrated, way off...

clhne, 2018-04-09:
Hi, Kris.
I have calibrated the mag, and the result look like this:
[image]
and I rotate the board, then I found a problem:
yaw angle is affected by pitch(when I rotate around the y axis), so why ?
Thank you.

kriswiner, 2018-04-10:
> Yes, I reference the data sheet:
> acc and gyro as:
>
> [image: icm20602 orientation]
> <https://user-images.githubusercontent.com/11884005/38530982-5464dd98-3ca0-11e8-9871-e6a121ca1061.png>
>
> mag :
> [image: ak8975 orientation]
> <https://user-images.githubusercontent.com/11884005/38530998-628720a2-3ca0-11e8-87ac-6e82b8a1f3d7.png>
>
> and the fusion filter as:
> update(gyro[1], gyro[0], gyro[2], acc[1], acc[0], acc[2], -mag[1], mag[0],
> -mag[2]);
> What may be the problem?

This is not an MPU9250...

clhne, 2018-04-10:
Yes, it's a 6-axis motion tracking device that combines a 3-axis gyroscope, 3-axis accelerometer, along with AK8975 3-axis electronic compass.
so Mahony fusion filter can be used, isn't it?

kriswiner, 2018-04-10:
both Madwick and Mohony filters take accel,gyro,mag input. Unless you
rerranged the functions, you are feeding gyro first...

kriswiner, 2018-04-10:
both Madwick and Mahony filters take accel, gyro, mag input. Unless you
rerranged the functions, you are feeding gyro first...

How are the mag and accel/gyro aligned then?

clhne, 2018-04-10:
The accel/gyro and mag aligned as: 
[image]

kriswiner, 2018-04-10:
Then this can't be correct... update(gyro[1], gyro[0], gyro[2], acc[1],
acc[0], acc[2], -mag[1], mag[0], -mag[2]);

Needs to be NED

clhne, 2018-04-10:
What's the order ?
update(-gyro[1], gyro[0], -gyro[2], -acc[1],acc[0], -acc[2], mag[1], mag[0], mag[2]); right?

kriswiner, 2018-04-10:
N E D

clhne, 2018-04-10:
Is the order update(-gyro[1], gyro[0], -gyro[2], -acc[1], acc[0], -acc[2], mag[1], mag[0], mag[2]) in accordance with NED ?

clhne, 2018-04-10:
The Mahony algorithm expects void MahonyAHRSupdate(float gx, float gy, float gz, float ax, float ay, float az, float mx, float my, float mz); you can find in:
http://x-io.co.uk/open-source-imu-and-ahrs-algorithms/ 
in "C code (Header and source files)"
So, why your function expects accel/gyro/mag ?

kriswiner, 2018-04-10:
Just a matter of convention; it simply depends on which mahony function you
are using. I only mention this because it is different from mine, so if you
were using my sketch then it would be wrong.

clhne, 2018-04-11:
Yeah, I use the source code not yours, and I reference the NED tutorial that you commented at : https://github.com/kriswiner/MPU9250/issues/257
I wonder if  (gx, -gy, -gz, ax, -ay, -az, -mx, -my, mz) is ok ?

closed 2018-04-23 by clhne