kriswiner/MPU9250 #263: Creating a 3D simulation using the MPU9250_BMP280_BasicAHRS_t3.ino sketch

maxerbox, 2018-04-08:
Dear,

I'm a highschool student, and I have to make a robot as a final exam project.
I'm in charge of the robot's orientation & Bluetooth transmission.
So we bough an [IMU 10DOF Grove](http://wiki.seeedstudio.com/Grove-IMU_10DOF_v2.0/)from seed studio, which works with a MPU9250 device. I have searched  "How can I get the orientation using an IMU board?", and discovered the Madgwick filter (I don't understand the maths, it's beyond me)

In order to get the orientation, I forked one of your skecth, [MPU9250_BMP280_BasicAHRS_t3.ino](https://github.com/kriswiner/MPU9250/blob/master/MPU9250_BMP280_BasicAHRS_t3.ino).

I have made some changes to the Arduino Sketch:
+ In order to receive the orientation & some infos, the arduino board print serial normalized infos (Yaw, pitch, roll.
+ I've made a new var for the informations printing interval. 

For the 3D simulation part, I'm using processing, [Processing](https://processing.org/), a visual art language based on java.

Here is a video of my experiment : 
[image]

The source code (Arduino & processing sketch) is available at : https://github.com/RISE-Project-STI2D/3D-Orientation-IMU-10DOF-MPU9250-Test-Grove

__**It is actually working, my problem is:**__
I don't get the same orientation as my display.
It's sort of inverted, when I rotate the IMU in a direction, on the display it rotates to another direction
What I'm looking for, is just to synchronize the IMU orientation with the display, and make them correspond when you rotate, like [this video](https://www.youtube.com/watch?v=BXsGWoOMtmU).

I was wondering if it was related to a NED frame, or something else.
Do you have an idea on the problem ?

kriswiner, 2018-04-08:
> IMU-10DOF-MPU9250-Test-Grove

No idea.

I would first make sure the heading pitch and roll makes sense on the board
output. If you point the board toward North, do you see heading of ~0
degrees? if you then point it Eastward, do you see 90 degree?

If you face the board up, do you read 0 degrees for pitch and 180 degrees
for roll? When you tip Eastern part of the board upward, does roll become
more positive?

When you dip the North end of the board downward, does the pitch become
negative?

Simple tests to see if you have the AHRS implemented properly on the board.
If so, then yes, there could be a mismatch between the coordinate systems
of the board and the display.

maxerbox, 2018-04-09:
>If you point the board toward North, do you see heading of ~0
degrees? if you then point it Eastward, do you see 90 degree?
 I'm not always getting the same value. ~~If I manage to get 0° heading  by orientating the board, and then turn to it by 90°, I get ~130° (I should get 90° normally ?).~~ I have remade a test and I get an angle of 80° by rotating it to 90°. That's unstable.

>If you face the board up, do you read 0 degrees for pitch and 180 degrees
for roll?
**If I face the board up, I get a pitch of ~0°, and a roll of ~0°**
**If I face the board down, I get a pitch of ~0°, and a roll of ~180° (it seems inverted)**
>When you tip Eastern part of the board upward, does roll become
more positive?
**Roll become more negative**
When you dip the North end of the board downward, does the pitch become
negative?
**Yes**

kriswiner, 2018-04-09:
This all seems fine to me, maybe there is a mismatch in your graphics
routine.

maxerbox, 2018-04-29:
Basically I have to align my display with the North to get matching
positions, is that right ?

kriswiner, 2018-04-29:
> issuecomment-379720000
> wUoK3bchYW3Hkl3Y8tk6vks5tm0VpgaJpZM4TLj9f

Sounds like it might be.