kriswiner/MPU9250 #264: Magnetometers Calibration issue

Luc-Panda, 2018-04-09:
Hello,

I use MPU9250 with a RFduino chip as BLE transmiter and host processor. 
I have some issues and question with magnetometers calibration.
To calibrate them I make my IMU rotate as a top spin around the three axis then I estimate the bias and gain by ellispoid fitting. The algorithm is effective but I observe that the different ellipsoids acquired shift a lot in a few laps time (the gain as well but with less effect).
I produce below acquisitions made on a two day period.
[image]

Does anybody as the same behaviour or an idea about it?

An other question, I don't find any subject about calibrating gyroscope gain. Are they steadily and effectively calibrate by the manufacturer?

Thanks for your help and your answers.

Best regards,

Luc Thomas.

Luc-Panda, 2018-04-10:
Sorry I wasn't very clear but I make the IMU turn around an axis and move the axis in the 3D space in order to catch maximum orientation. I repeat that with 3 orthogonal axis.

As you can see on the figure above, I catch a sufficient number of point in all the 3d space to obtain very nice ellipsoids.

My issue is that the ellipsoids are not steady in time. You can see that the orange acquisition is clearly translated from the other ones.  It's like the bias has changed on a two days period. Is it usual behaviour?

I try to understand why in order to avoid this. In fact, I'd like to make one calibration for "ever" and not have to do it each time I want to use the IMU.

With the acquisition of the figure above, I obtain:
bias (x, y, z):
[  2.06229712  14.22784194  18.14341732]
[ -5.97952045  26.2406433   39.8837559 ]
[  2.84221251  13.97562296  16.58551901]
[  2.61622145  13.94951021  16.99077708]
[  2.61010596  13.87354858  16.83966887]

gain matrix:
[[  1.24095724e+00  -1.48399795e-04   2.49767903e-03]
 [ -1.48399795e-04   1.20843524e+00   5.30881510e-02]
 [  2.49767903e-03   5.30881510e-02   1.12661415e+00]]
[[ 1.20829074  0.00162151 -0.01143995]
 [ 0.00162151  1.26426121 -0.00150996]
 [-0.01143995 -0.00150996  1.16574819]]
[[  1.25033527e+00   5.23471561e-04   2.24227668e-03]
 [  5.23471561e-04   1.28703885e+00  -2.81913511e-03]
 [  2.24227668e-03  -2.81913511e-03   1.21971159e+00]]
[[  1.23023192e+00   3.81163190e-03  -1.18164347e-03]
 [  3.81163190e-03   1.24822284e+00  -1.70080953e-02]
 [ -1.18164347e-03  -1.70080953e-02   1.15243108e+00]]
[[ 1.21040627 -0.00695667 -0.00422019]
 [-0.00695667  1.24840543  0.00208103]
 [-0.00422019  0.00208103  1.14282565]]

Thanks for your answers.

kriswiner, 2018-04-10:
The sensors will be sensitive to changes in temperature and other
environmental conditions so I doubt you will be able to have one
calibration for all time, at least for the mag.

Luc-Panda, 2018-04-20:
Hello Kris,

I made electronical test and the magnetometer bias is steady with power supply change and different current variation in the PCB.

But in fact, the biases are highly related with PCB mechanical stress. I use a MPU 9250 with a AK8963 magnetometer. Is there any solution to avoid this sensibility by special PCB mounting operation or do I have to change the magnetometer and so with which one?

Regards,

Luc.

kriswiner, 2018-04-20:
All mems sensors show strain-induced effects, that is wht they need to be
calibrated.