kriswiner/MPU9250 #265: Inestability oh the yaw

RicardPalo, 2018-04-10:
Hello Kris,
I'm doing a project about a self guided robot car, so I use the MPU9250 as compass. I have followed your codes and calibration advices. 
The problem is that after calibrating the MPU9250 I obtain good results, but only on the point I've made the calibration. When I calibrate the sensor, the robot car is stopped and the yaw is good. But when the car moves to another position (because the car has to follow a path), sometimes the yaw gotten is really different than what it should be. 

I suppose that it is happening because of another magnetic fields. When I move the car keeping it 1.5 metres away from the ground, the yaw remains good. Maybe there ara magnetic fields under the ground generated by wiring or something?

I'm a little lost, because I don't know what I can do. Maybe make the calibration in different points where I know the car will pass through, and then take the magbias averages?  

Please, tell me what do you think about it. I would be grateful if you could give me any advice or solution.

Thank you so much for your time.

kriswiner, 2018-04-10:
If there are variable, external magnetic fields present from otors/wires
whatnot the best you can do is use the accelerometer to detect and correct
these magnetic field anomalies. it would take a bit of doing to come up
with an algorithm for doing so but this might be a fun task. We generall
use the EM7180
<https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/>
to avoid this issue, which has magnetic anomaly detection/correction built
in. So maybe this would be an easier solution?

RicardPalo, 2018-04-11:
If I want to use the accelerometer to detect and correct magnetic field anomalies, where should I start from??

I think that maybe, I could make something to keep the MPU9250 far enough from the ground to have good results enough.

kriswiner, 2018-04-11:
Not sure.