kriswiner/MPU9250 #266: MPU9250 with CC2650 magnetometer issues

ajmellers, 2018-04-11:
I would just like to start by thanking you for your contributions in regards to the use of the MPU9250.
We have been using the MPU9250 in conjunction with a cc2650 which on our chip we have always been able to use the gyro and accelerometer on the device, recently we realised the benefits of the magnetometer and began developing firmware for such, to no avail, we then bought the development board for the MPU9250 you use with the teensy, this worked fine with your code, then using the MPU9250 board with the CC2650 then works fine with the code we had completed previously for our own chip, from this it is very difficult to decipher the issue as such, the chip is wired exactly as  on the development board, are there instances of the AK8963 not working or can you advice use on what the issue maybe?

thanks again,
Alex

ajmellers, 2018-04-11:
just to put that a little less wordy:
custom board : gyro and accelerometer works fine magnetometer doesnt communicate
Launchpad and mpu9250 development board : all work fine
Teensy and MPU9250 dev board: all work fine

and it is definitely a mpu9250 and not just the 6250,

thank again.

kriswiner, 2018-04-11:
Not sure what to say here. It seems the software works fine. There is no
reason why the mag in the MPU9250 wouldn;t work as long as you are using
the same software. Can you at least see the mag (0x0C) on the I2C bus when
you do a scan? Normally, all that is required is to enable bypass mode in
the INT_CFG register of the MPU9250 and this allows access to the mag.

You say a custom board. Is it possible the connections are faulty? You do
have to connect certain pins to 3V3 to get the mag to work, like pin 1 and
pin 22 IIRC, etc. So could be a hardware problem on your custom board most
likley.

ajmellers, 2018-04-11:
Thanks for the quick reply, no the who am does not work on the AK8963 (does on the evaluation board) both 1 and 22 are tied to VDD, yeah I think so too just not sure where!

kriswiner, 2018-04-11:
But if you scan the I2C bus for devices, does the AK8963C appear at 0x0C?
The who_am-I is 0x48, so if the auxiliary I2C bus is not enabled you won't
see the mag nor be able to read from it. Did you enable the pass through
mode using INT_CFG register?

What does the top of the MPU9250 pachage say? It should say MP 92...

ajmellers, 2018-04-12:
Yeah, it says MP 92, at a loose end! thinking we may just use these boards with accelerometer and gyroscope, and move on to another chip, what would you advise? (apologies for late reply i feel we may be on different times!)
Thanks again,
Alex

kriswiner, 2018-04-12:
It makes no sense that the mag doesn;t work on your board. I have never
heard of such a thing after assembling hundreds of MPU9250.

You could use the LIS2MDL magnetometer from ST; it is a fine sensor as is
their LSM6DSM accel/gyro and LPS22HB barometer.

If you send me the schematic portion with just the MPU9250 I might be able
to find the error? One thing is to have no copper under the MPU9250 package
and no thermal pad or solder on the pad. But this has to do with mag
performance, and shouldn;t affect whether the mag shows on the I2C bus.

ajmellers, 2018-04-12:
here is the trace of the MPU, on the right, no pad or solder underneath

kriswiner, 2018-04-12:
> issuecomment-380688500

Not seeing any attachments. I don;t think you can attach in github. Try
tleracorp@gmail.com