kriswiner/MPU9250 #268: Load DMP firmware to MPU9250

PauwelsKristof, 2018-04-18:
Hi,

I am working on a project to calculate the yaw, pitch and roll angles via a MPU9250. For that i want to use the DMP for the Low Power Quaternion. But i want to use as little libraries as possible to get a good understanding of what is really going on.

In the 'Programming Sequence for DMP Hardware Functions' application note i've found how to enable the advanced hardware features at page 35 but at the end i need to load the firmware into the DMP. Do you know where i can find the firmware or is there a default firmware present on the chip?

Best regards

kriswiner, 2018-04-18:
There is no firmware on the MPU9250, at least not sufficient firmware. You
have to download a ~5 kByte binary blob into the chip using the MCU. Ask
Invensense where to get this.

BTW, the DMP is opaque, you will undertsand nothing by using it. Also, it
runs at a fixed 100 Hz and doesn't use the mag data, so Yaw will be subject
to rather wicked drift. See here:

https://github.com/kriswiner/MPU6050/wiki/Affordable-9-DoF-Sensor-Fusion

Looks like ~0.4 degree per minute.

PauwelsKristof, 2018-04-18:
Thanks for the quick reply!

So is the best way than just reading the raw data and afterwards on a seperate MCU calculating the pitch, yaw, roll via a Madgwick filter? Or what would you suggest?