kriswiner/MPU9250 #269: How to obtain a certain angle from MPU 9250 sensor (specifically gyro)

gdt96, 2018-04-19:
Hi,
We want to ask about how to obtain a certain angle from MPU 9250 sensor. Our project is about an air drum that function based on arm movement. But the problem we face now is we cannot get the exact angle when we move the sensor. Specifically, we only use gyroscope part on this sensor. Is there any calculation to obtain the angle? Because when we move the sensor to a certain angle, the angle that we got is not logic and unacceptable. Can we have your explanation on this problem?

kriswiner, 2018-04-19:
You can't do this with just the gyro, this is the whole point of a 9 DoF
sensor, use all of thesensor data to get absolute orientation, then you can
get an accurate angle for any location in space..

gdt96, 2018-04-22:
Hi kris,

can you suggest multiplexer to multiplex MPU 9250 sensor. (to multiplex the data not clock) We tried to use 74LS151 as multiplexer but there is no output

kriswiner, 2018-04-22:
If you have access to the ADO pin of the MPU9250 breakout boards you are
using you don't need a multiplexer. Just toggle all of the addresses to
0x69 (set ADO==HIGH), set the one you want to read from to 0x68 (ADO==LOW),
read the data, reset to 0x69 (ADO==HIGH) and move on to the next one, etc.

Or you canuse an I2C multiplexer, I like the TCA9548A
<https://www.tindie.com/products/onehorse/tca9548a-i2c-multiplexer/>.

gdt96, 2018-04-22:
We have tried to connect AD0 pin of the MPU9250 but our main problem is on the coding. Right now, we connect 2 MPU 9250 sensors to the multiplexer but we dont really know how to make the coding so that Arduino can read both sensor at the same time. Can you show some coding examples for this problem?

kriswiner, 2018-04-22:
If you only have two, forget the multiplexer and use the ADO, one set high
and one set low.