kriswiner/MPU9250 #272: Why is NED mapping different for mpu9250 examples

ifrew, 2018-05-01:
Hi Kris,

First off thanks for all your work in this area!

I'm having a hard time trying to.get my head around why in your examples the mappings for NED change for the the same 9250 chip. Currently using your mpu9250-5611 code for the breakout board Im using. If the x axis points forward away from you that is North, correct? Based on your comments in the code and in the datasheet, the x-y accel and gyro axis are aligned with the y-x of the magnetometer  and the + Z axis of the mag points down while the accel points up.

Given that, why wouldn't the mappings then stay the same for each example. 

For example, in the 9250-5637 example you have 

// Sensors x (y)-axis of the accelerometer/gyro is aligned with the y (x)-axis of the magnetometer;
  // the magnetometer z-axis (+ down) is misaligned with z-axis (+ up) of accelerometer and gyro!
  // We have to make some allowance for this orientation mismatch in feeding the output to the quaternion filter.
  // For the MPU9250+MS5637 Mini breakout the +x accel/gyro is North, then -y accel/gyro is East. So if we want te quaternions properly aligned
  // we need to feed into the Madgwick function Ax, -Ay, -Az, Gx, -Gy, -Gz, My, -Mx, and Mz. But because gravity is by convention
  // positive down, we need to invert the accel data, so we pass -Ax, Ay, Az, Gx, -Gy, -Gz, My, -Mx, and Mz into the Madgwick
  // function to get North along the accel +x-axis, East along the accel -y-axis, and Down along the accel -z-axis.
  // This orientation choice can be modified to allow any convenient (non-NED) orientation convention.
  // Pass gyro rate as rad/s
    MadgwickQuaternionUpdate(-ax, ay, az, gx*PI/180.0f, -gy*PI/180.0f, -gz*PI/180.0f,  my,  -mx, mz);

And this is different from the 9250-5611 example. I hope you can elaborate as im getting wrong results in my code. Also, why are the gyro values negative for y and z. Rotational direction around the axis wouldn't change with an accelerometer axis sign change?

Of all the coding I've done with these devices this ned mapping has always confused me.

Cheers
Iain

ifrew, 2018-05-01:
> Thanks for quick reply Kris. That makes sense. I always thought it was determined by the silkscreen on breakout board.
>
>
> The user selects which edge of the board is to be defined as North, then
> everything else floows from that. This is why different users might have
> different definitions of NED.

Oh. Forgot to ask.. Is gravity component already removed in the accelerometer readings returned by your library. I don’t think so looking at your code but wondered if the IMU was doing that  but wanted to check as I am rotating the vector returned (ie IMU.ax,ay,az) to subtract it in the vertical axis (I’m only interested in the vertical component) but getting values of 0 for az and when I then turn the device upside down a reading of 2.0 (ie 2g)

kriswiner, 2018-05-01:
> safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.
> com%2Fkriswiner%2FMPU9250%2Fissues%2F272%23issuecomment-
> 385746402&data=02%7C01%7C%7C8f8ca70371164712ddae08d5af8ff9a3%
> 7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C636607955251424031&sdata=
> x4OlwTLdZoN5dFYuS0whiDB5WmiT5ks5tuKcEgaJpZM4TuHNF&data=02%7C01%7C%
> 7C8f8ca70371164712ddae08d5af8ff9a3%7C84df9e7fe9f640afb435aaaaaaaa
> aaaa%7C1%7C0%7C636607955251424031&sdata=PuFkaJzer66Cw9y%

Gravity is not removed in the standard ax/y/z scaled output. There is a
function to do so in some of the later sketches by constructing the
rotation matrix and forming linear acceleration (with gravity removed).

ifrew, 2018-05-01:
Thanks Kris. I calculate that already, just wanted to be sure I wasn;t taking it off twice!.

Interestingly I noted I had the magcal odr rate set wrongly for calibration. It was 2 so I set it to 6 100hz same as for the accel gyro rates and it came out it came out at approx. 1 and -1 when I waved in figure of 8. However, doing a reset it went back to accel readings of 0 and 2 again when upside down. After resetting each time and looking at the results I am seeing either these 2 types of data being returned approx when the device hasn’t moved (readings are in G’s)

ax 0.16 ay -0.50 az -1.19
Raw Ax: 0.16 Raw Ay: -0.50Raw Az: -1.18

Or

ax 0.85 ay 0.06 az -0.00
Raw Ax: 0.85 Raw Ay: 0.06Raw Az: 0.01

Saw other people reporting different results on the forum about the mpu9250. Duff device do you think? Just soldered it up yesterday! 😊

BTW, where do I send the beer money! Seriously. Your contributions on sensors have been very educating.

Cheers
Iain

<mailto:MPU9250@noreply.github.com%3cmailto:%0b>> MPU9250@noreply.github.com<mailto:MPU9250@noreply.github.com>>>

ifrew, 2018-05-01:
> No the device is fine, pilot error.
>
> The accel calibration is designed for one particular orientation, MPU9250
> pointed up. If you point it down, you will be adding 1 g to the az instead
> of subtracting.

Lol.. Funny because I’m a paragliding pilot and it’s a vario I’m developing!

In my case the mpu9250 is upside down in the case, so what would be your suggestion for me to get the right values other than calibrate it the right way up.
Would I have to change the NED mappings  to the right in the calculation or simply subtract 2g from the z value. I thought I could just keep the NED mapping the way it is for the right way up, have the device upside down but then change the sign on the results for z acceleration. I’m only interested in the z component. Sounds like I can do that but subtract 2g’s.  Sound about right?

Cheers
Iain

<mailto:reply@reply.github.com%3cmailto:%0b>> reply@reply.github.com<mailto:reply@reply.github.com>>>; kriswiner/MPU9250 <
<mailto:notifications@github.com%3cmailto:notifications@github.%0b>> com<mailto:notifications@github.com%3cmailto:notifications@github.com>>>
<mailto:MPU9250@noreply.github.com%3cmailto:%0b>> <mailto:MPU9250@noreply.github.com%3cmailto:%0b>>
<mailto:ifrew@hotmail.com%3cmailto:ifrew@hotmail.com%3cmailto:ifrew@%0b>> hotmail.com%3cmailto:ifrew@hotmail.com>>>; Author <
<mailto:notifications@github.com%3cmailto:%0b>> <mailto:notifications@github.com%3cmailto:%0b>> notifications@github.com<<mailto:notifications@github.com%3c>
<https://nam03.safelinks.protection.outlook.%0b>> <https://nam03.safelinks.protection.outlook.%0b>>

kriswiner, 2018-05-01:
Yes, you can fix the accel cal function for your preffered orientation or
just calibrate with the sensor pointed up. No need to change the ned
assignment.

ifrew, 2018-05-01:
Many thanks Kris..got it showing +1 and -1 now each way...I'll close this now.

cheers

Iain

closed 2018-05-01 by ifrew

eminyavuzaslan, 2024-08-10:
Hi @kriswiner ,

i dont know where should i write my problem so i decide to write here because my problem is same as here i guess.

i have a mpu9250 board and i place this IMU in a model rocket vertically.

So my accel gyro and mag axes like this ;

[image]

and  i map the sensor values to ned like this

MadgwickQuaternionUpdate(az, ax, ay, gz*PI/180.0f, gx*PI/180.0f, gy*PI/180.0f, -mz, my, mx);

Is it true ?

and i have another question

I told that i place the imu in a model rocket vertically so when rocket nose cone through the north mpu9250 is not pointed up. How should i calibrate it when imu is in the model rocket ?

kriswiner, 2024-08-10:
Looks like the sensor mapping is correct for NED.

eminyavuzaslan, 2024-08-11:
Thanks for fast reply

So what should i do for calibrate the mpu when inside of the rocket ?

I told that i place the imu in a model rocket vertically so when rocket nose cone through the north mpu9250 is not pointed up. How should i calibrate it when imu is in the model rocket ?

kriswiner, 2024-08-11:
No idea. You can calibrate when outside of the rocket and check it returns
the expected results. I suppose you can place it in the rocket and, if you
have some telemetry or can attach a serial port you can verify calibration
inside the rocket.

eminyavuzaslan, 2024-08-15:
When i calibrate it when outside raw accel values are not correct. I told that i place the IMU vertically so my accel and gyro axes are y is down z is north and x is east and my outputs when NED z is down  0 .3- 9.85 - 0.2 . It has to be like 0-0-9.81.

 So according to this answer;

https://github.com/kriswiner/MPU9250/issues/272#issuecomment-385795100

What should i change in Calibration code the correct accel values ?

thanks

kriswiner, 2024-08-15:
"my outputs when NED z is down 0 .3- 9.85 - 0.2 "

accel outputs?

The NED is for the Euler angles, not the acceleration. If you have the
y-axis down you will see 9.81 on the y-axis of the accel.

eminyavuzaslan, 2024-08-16:
Hi again Kris

Yes, my accel outputs ,

I'm just printing the accel value that i mapped the ned

You approved that this code line ;

// MadgwickQuaternionUpdate(az, ax, ay, gzPI/180.0f, gxPI/180.0f, gy*PI/180.0f, -mz, my, mx);

you can see my x accel is az, y is ax and z is ay

when imu vertically according to Accel and Gyro axis y is looking down so yes i have to see 9.81 but when imu vertically according to my NED axis y is not looking down (https://github.com/kriswiner/MPU9250/issues/272#issuecomment-2280375282) So if i place the imu vertically i have to see 9.81 in z axis because i define the NED like that . I'm avare of i have replace accel values like NED mapping to see true accel values so I'm printing accel values with this code down below;

  Serial.print(accel[2]);
  Serial.print(" ");
  Serial.print(accel[0]);
  Serial.print(" ");
  Serial.print(accel[1]);
  Serial.println(" ");

Now if i do everything correct , when imu vertically i have to see 9.81 for accel[1] (z axis according to my NED) 

But i can't. I think it's a calibration or pilot error.

You told that (https://github.com/kriswiner/MPU9250/issues/272#issuecomment-385795100 and https://github.com/kriswiner/MPU9250/issues/272#issuecomment-385821827)  we have to calibrate accel when mpu is pointed up. In my project mpu is not pointed up what should i change in calibration part ?

Huge thanks for replying my dumb question

Best Regards

kriswiner, 2024-08-16:
> is not looking down (#272 (comment)
> You told that (#272 (comment)
> and #272 (comment)

In the case that the MPU9250 is pointed down instead of up, the z-axis is
still sensitive to gravity and one can correct the reported value by a
simple scaling.

In the case that the y axis is pointed down, there is no simple scaling
that can result in the z-axis being sensitive to gravity.