kriswiner/MPU9250 #273: How to make graph in Calibration tutorial?

Patrickyp, 2018-05-10:
I am looking at this tutorial and I don't know how the dotted graphs were made.  What exactly is the format of the magnetometer to print to serial monitor and get that graphed data?  Any example code?  

tutorial:
https://github.com/kriswiner/MPU6050/wiki/Simple-and-Effective-Magnetometer-Calibration

Thanks.

kriswiner, 2018-05-10:
> Effective-Magnetometer-Calibration

Capture mx, my,mz to serial monitor, import into spreadsheet, plot mx vs
my, mx vs. mz, my vs. mz.

Patrickyp, 2018-05-11:
Thanks for your answer.  So should I be copying your magcalMPU9250() function in to my code?  I am running MPU9250BasicAHRS.ino which is lacking that test but has the function calibrateMPU9250() which I assume alone is insufficient?

kriswiner, 2018-05-11:
Yes, you wil need some way to calibrate the mags. Why not start with a more
modern sketch altogether like, maybe, this one:

https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_BasicAHRS2_ESP8266.ino

Patrickyp, 2018-05-15:
Would MPU9250BasicAHRS.ino work just as well assuming I implement the magcalMPU9250 function correctly?  I am having trouble getting the one you linked to compile as I am not using MS5637 just mpu9250 from sparkfun and a teensy 3.6.

kriswiner, 2018-05-15:
Maybe, but not the latest...

Patrickyp, 2018-05-21:
DO you know how often I need to do the figure 8 calibration?  It will be a hassle for the project I'm working on but it seems like I have to do it pretty frequently to get accurate data.

kriswiner, 2018-05-21:
Just do it once, well, and store the values in the Arduino sketch so you
don;t have to do it again, or only rarely.

MitalPattani, 2018-07-22:
Try using serial plotter instead if just want to check the calibration.

khoa21214475, 2018-07-22:
I will try, thanks ! 😃

zenmanenergy, 2018-07-22:
If you get that to work, will you please post a link to the plotter you used and/or a short video of getting it working? I'm struggling with calibration too.

kriswiner, 2018-07-22:
Stream comma delimited mag data to the serial monitor, grab it and import
into any spreadsheet to plot Mx vs. My, My vs. Mz, Mx vs. Mz, etc.

Not sure why this is so hard...

khoa21214475, 2018-07-24:
may be i have good chart...
[image]
Thanks Kriswiner so much !

khoa21214475, 2018-07-24:
And an additionally question, how can we calibrate accelerate and gyro sensor, put face up on flat and  stand still, right?

kriswiner, 2018-07-24:
Yes, or with a little more effort like this
<https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/F.--Magnetometer-and-Accelerometer-Calibration>

khoa21214475, 2018-07-25:
I can't buy em7180 + mpu9250 + bmp280 😂.
2 degree with this board, and how about the accuracy without em7180 and bmp280 ?

kriswiner, 2018-07-25:
" I can't buy em7180 + mpu9250 + bmp280 "  Why not?

Without EM7180 you could get 4 degree heading accuracy when well
calibrated. See this
<https://github.com/kriswiner/EM7180_SENtral_sensor_hub/wiki/K.-Limits-of-Absolute-Heading-Accuracy-Using-Inexpensive-MEMS-Sensors>

khoa21214475, 2018-07-25:
Thanks for quickly replies Kriswiner!
I will try to get one module with ESP32 + GPS Neo 6M + mpu9250 + bmp280 + em7180 , i want to build this module for tracking celestial body.
Im Ok to get Az, Alt from module GPS Neo 6M, and now i will use two information to control step motor by using mpu9250 + bmp280 + em7180.
But its not easy to get minimum accuracy with mpu9250 😅