kriswiner/MPU9250 #274: temperature sensitivity?

mpava, 2018-05-22:
Hi Kris,

First off, thanks for the great material and forum you have created! :)

I am tinkering with MPU9250 and am having great success with the madgwick algo calculating the orientation of the sensor. What slightly spoils it beauty though is the error which could be up to 30 degrees in some orientations. I narrowed this down to calibration. If I recalibrate just before the use, the error reduces to about 5 degrees. My theory is that it's the temperature that affects the mag biases. The mag calibration sphere seems to move about 10 percent when temperature changes about 10C. It seems that a relatively small movement of the calibration sphere has quite visible impact on the quality of the orientation. Have you experienced this behaviour yourself? If it is the temperature indeed, than modifying the algo to compensate for this could be quite fiddly. Any ideas?

Also it seems that the calibration sphere can be slightly skewed, not massively. However I suspect the skewiness is contributing to the error. Any thoughts how to correct it?

What kind of margin of error do you get - actual orientation vs calculated one?

anyway thanks again!

cheers

Mike

kriswiner, 2018-05-22:
Mems sensors are basically harmonic oscillators and their behavior will
depend strongly on temperature. 10 degrees C is a huge temperature swing,
so I am not surprised all of the sensors have their bias calibration upset.

I don't think there is a general way to take this into account; you simply
have to have a provision for calibrating all of the sensors (especially
mag) from time to time or when conditions (like temperature) change.

There are more sophisticated methods (matrix mechanics) for correcting
elliptical variation in the mag response. We use them and it greatly
improves the heading accuracy. With simple offset bias corrections and
axial response scaling ~4 degree heading accuracy is about what you can
expect from the MPU9250 and Madgwick or Mahony fusion algorithms.

mpava, 2018-10-21:
Hi Kris,

Any thoughts re below would be much appreciated. I am still not fully happy with the accuracy I am getting from the madgwick algo. I keep getting up to 20 degree errors in some directions. I think it is related to the magnetometer.  To prove this, I rotate the sensor in 90 degree increments around each axis (x,y,z). This produces a data set which looks like 3 circles with clusters of measurement points in XY, XZ and YZ panes respectively. When I overlay the circles on top of each other into a 2D graph, I would expect them to perfectly match. I however get some of the clusters way off where they should be. See attached graph (each plane in different colour - the clusters in top-right quadrant are way off from each other, the others are a lot better):

[image]
I think the misalignment of the mag data is the root cause of the issue. On some occasion I managed to get the perfect alignment of the mag chart above, which lead to perfect output from madgwick. I can't seem to reproduce the good result or get consistency between measurements even if the conditions don't change. I also noticed that the sensor is sensitive to pressure on PCB, my board is quite thin and so if I put a little pressure in parts of the board the misalignments above change positions; slight temperature changes can do the same.

I am going to make a new board which will be a lot thicker to avoid any PCB surface pressure propagate to the sensor and also give the MPU ample amount of space away from vias, traces etc. Currently my layout is quite tight, although within recommendations - i.e. no traces under the MPU etc, some vias and I2C lines are quite close to the MPU though. I haven't soldered the middle pad etc.

I use the calibration method you describe in your blog where I offset and scale the calibration data into a sphere. When I draw the calibrated data set I get a perfect sphere as per below: 
[image]

So in summary, the perfect calibration sphere as per above doesn't yield perfect measurements of the actual rotations.

any thoughts would be welcome :)

thanks

Michael