kriswiner/MPU9250 #275: Delay required for bypass enable?

Ironicster, 2018-05-24:
Hello kriswiner
First of all thank you for this great library!
Not sure if this should be listed as a issue.

I adapted you library to use two MPU9250's (sparkfun verison)
I am switching the I2C slave bypass enable alternately between each chips to be able to access the magnetometer separately. However it seems that some times i get readings from the wrong one.

If i add a delay(2 or more ms), this no longer occurs, but at the cost of having lower frame rate.
Do you know if there is some sort of settling time between setting the bypass enable and being able to read from the chip. Could not find anything on the datasheets.

Regards

kriswiner, 2018-05-24:
I don't think this is noted in the data sheet but others have found 1 or 2
ms is needed to get the switch to work robustly, so I am afraid this is the
price of this solution. You could use the MPU9250 as I2C master and read
the mag data into the I2C slave registers, then you would be able to read
all of the data from the MPU9250 directly. This is a little more
complicated to set up and I doubt the response rate would be much faster,
since you would have to wait for the mag data to be read over the MPU9250
master I2C bus anyway. Likely, the method you are using is about as well as
can be done without an I2C multiplexer and the ~500 Hz limitation is
something you are just going to have to live with.

Given that the max rate of the mag is 100 Hz anyway, you could read the mag
only every other pass through the loop (if you are using 200 Hz for the
gyro and accel) to reduce wasted time. The latency introduced is reduced
further as you increase the accel/gyro sample rates. 1 kHa isnot an
unreasonable rate for these snesors for a lot of applications. So the time
lag can be managed, but not eliminated.

Ironicster, 2018-05-24:
Thank you for the confirmation.
I went up to 5ms delay. Event at 3ms i would occasional get a spurious reading. 
I am using this for a gesture tracking application so i guess the framerate does not need to be that high. It is annoying though having to add a delay on the code.

where should the case of beer be sent?

kriswiner, 2018-05-24:
No beer necessary this time...