kriswiner/MPU9250 #276: After bypassing I get 3 sensors on I2Cscan

Saroro, 2018-05-24:
Hello, 
So I am using 2 slaves by bypassing sensor simultaneously but on bypassing the sensor 0×68,  in I2Cscan I get 0×68, 0×69, 0×0C sensors, is that okay? Shouldn't I be getting only 0×69 only??

Saroro, 2018-05-24:
Then how do address both 0×0Cs separately in passthrough mode??

kriswiner, 2018-05-24:
Enable passthrough mode for one, disable for other, then switch, etc

Saroro, 2018-05-24:
I did that thing that first time! That's what I am saying
{
writeByte(MPU9250ADDRESS_1, INT_PIN_CFG, 0x12);
writeByte(MPU9250ADDRESS_2, INT_PIN_CFG, 0x10);
}
else{
writeByte(MPU9250ADDRESS_1, INT_PIN_CFG, 0x10);
writeByte(MPU9250ADDRESS_2, INT_PIN_CFG, 0x12);
}

kriswiner, 2018-05-24:
try adding a 5 millisecond delay, it takes some time for the switch to
engage

Saroro, 2018-05-24:
I did! I added 5us in both loops even 10us still doesn't work, I only read one sensor

Saroro, 2018-05-24:
Passthrough mode fuse set is also reads FF

kriswiner, 2018-05-24:
5 milliseconds

Saroro, 2018-05-24:
Yess I tried 5milliseconds in both cases, the ak8963_cntl sensor reads B5

kriswiner, 2018-05-24:
> Even when I changed the value of ak8963_cntl t 1F

Not sure what to tell you, others seem to be able to make this work:

https://github.com/kriswiner/MPU9250/issues/275

Saroro, 2018-05-24:
But for this code, I should get only 2 addresses each time right??
Either 0×68 or 0×69 right??

{
writeByte(MPU9250ADDRESS_1, INT_PIN_CFG, 0x12);
writeByte(MPU9250ADDRESS_2, INT_PIN_CFG, 0x10);
I2cscan
}
else{
writeByte(MPU9250ADDRESS_1, INT_PIN_CFG, 0x10);
writeByte(MPU9250ADDRESS_2, INT_PIN_CFG, 0x12);
I2cscan
}

kriswiner, 2018-05-24:
Nope. You will get both 0x68 nd 0x69 if you have two MPU9250 on the same
I2C bus.

Bypass simply lets the MCU host talk directly to the AK8963C.

MitalPattani, 2018-05-24:
But here when I try to get the reading it's not bypassing the first sensor at all! I tried to simply bypass the sensor1 it still gives me acc values for sensor1! I am using arduino does that affect it in anyway?

Saroro, 2018-05-24:
I tried to just bypass the first sensor, I still reads the first sensor! Any suggestions?

kriswiner, 2018-05-24:
loop

read(0x68) // read device one

process accel1 data

enable bypass(0x68) //  open I2C switch for device 1
disable bypass(0x69) // close I2C switch for device 2

delay(10)

read mag data
process mag data

read (0x69) // read device 2

process accel2 data

enable bypass(0x69) //  open I2C switch for device 2
disable bypass(0x68) // close I2C switch for device 1

delay(10)

read mag data
process mag data

endloop

It's really not that hard.

AGAIN: BYPASS DOES NOT BYPASS THE MPU9250, IT ALLOWS THE MCU HOST TO TALK
DIRECTLY WITH THE MAGNETOMETER.

If you want the data only from one of the MPU9250, then you have to use its
I2C address, nothing to do with bypass mode.