kriswiner/MPU9250 #277: Libraries not getting added properly

shivams2u2, 2018-05-29:
Hello sir, 
I am a beginner and i have bought Arduino mega and arduino MPU-9250. I am trying to find roll, pitch and yaw without display but i am not able to add libraries properly i guess. 
I have copied these libraries which are being uploaded by you in arduino library folder. 

I have attached  the screenshot of the code and error.

[image]
[image]

kriswiner, 2018-05-29:
Why not just run the sketch as is?

You've added libraries here that are not part of the original sketch, I
can't help you with that...

ssammut95, 2018-05-30:
I don't think this has to do with libraries. Use the code Kris called "MPU9250BasicAHRS.ino" and remove all instances of the Adafruit screen if you are not using it.

ssamm95, 2018-05-30:
I don't think this has to do with libraries. Use the code Kris called "MPU9250BasicAHRS.ino" and remove all instances of the Adafruit screen if you are not using it.