kriswiner/MPU9250 #278: Tilt without Gforce

itayrosenml, 2018-06-03:
Hi,

I'm trying to detect a slope in a device (mobility scooter to be exact). Using Pitch, I'm able to get exact degrees of rotation. However, the values are affected by the Gforce and therefore when the device is in motion (e.g. driving, accelerating, slowing down), so does the Pitch. 

Is it possible to get rotation degrees around Yaxis which is not affected by Gforce?

kriswiner, 2018-06-03:
You could remove gravity from the acceleration and re-calculate quaternions
I suppose...or just use the linear accelerations to estimate pitch.

itayrosenml, 2018-06-03:
Hi,

Unfortunately acceleration is measured with Gforce, so I guess I need to somehow measure it differently.