kriswiner/MPU9250 #280: Madgwick update rate

Micro213, 2018-06-03:
Thanks for files and support
does it help to have Madgwick filter update rate at 50 khz or more?
i am using ESP32, and sensor update speed seems capped at around 847hz ~,  cpu is not the bottleneck for sure, can it be I2C speed?

MPU9250_SMPLRT_DIV  set to 0x00
MPU9250_CONFIG  set to 0x02
MPU9250_ACCEL_CONFIG2 set to 0x02

any information much appreciation to you

kriswiner, 2018-06-03:
Gyro sample rate or fusion filter rate is 847 Hz?

Micro213, 2018-06-03:
gyro sample rate 847hz

fusion is really high on a separate core

kriswiner, 2018-06-03:
Then I don't understand what you are asking me. If the gyro rate is 847 Hz,
then a fusion rate of 10 kHz is likly overkill. I would set the number of
iterations per gyro data sample as a variable and see what you can get away
with. 5, 10, 50? I typically un the gyro and accel at 1 kHz and mag at 100
Hz, then fusion rate can be determined by 10 or 20 iterations of the
Madgwick filter.

Micro213, 2018-06-03:
this is my question answered

Thanks

closed 2018-06-03 by Micro213