kriswiner/MPU9250 #281: MPU9250 will be placed in random position - is there a way to get it working properly ?

jarekkt, 2018-06-04:
Well, maybe not completely random, but still it will not be aligned to XYZ planes.

I have a device which my clients can attach to their boats in semi-random location and orientation.
This means that  I need to be ready to work even if e.g. the MPU 9250 XYZ planes are tilted.
I can make factory calibration to fix base offsets ( for gyro/magnetometer/acceleremeter).
User, after installation also makes single manual calibration ( so I know the installation orientation).

But then - what to do ?  While gyro might not be affected, but magnetometer and acceleromter calibration does not seem to be trivial anymore. Especially that  I cannot rotate the device anymore .

Has anyone developed runtime calibration for the MPU9250 ?

kriswiner, 2018-06-04:
This is confusing.

Sensor fusion outputs quaternions from well-calibrated 9 DoFsensor data.
The quaternions can be used to estimate absolute orientation of the device
wrt gravity and magnetic north independent of any mounting considerations.

Are you not calibrating the sensors?

Are you not using sensor fusion to estimate absolute orientation?

kriswiner, 2018-06-05:
> Well, I do calibrate it - during production. Once.
>
> But then the device is installed and its position is fixed in semi -
> random position.
> The position estimation work, is quite precise (both roll and pitch),
> having GPS also helps to adjust true heading.
>
> I keep original calibration data in non-volatile memory, but it is not
> sufficient in long term. I can see that sensor heating takes its toll and
> it starts drifting. From tests you need about 5 minutes until it reaches
> stable point.
> I am also not quite sure about sensor aging - my device is rated for 10
> years of lifetime and I have to be sure it is compensated
>
> So I would like to recalibrate it. I cannot do it directly after power-on
> as it is not anymore in base position so I cannot simply take offsets
> directly from measurements.
>
> I have asked about runtime calibration - DNP computing seems to detect 'no
> movement' and adjust zero offsets on the fly. In theory checking variations
> of measurements should detect standing still - especially for gyro. I
> cannot do much for magnetometer - during normal operation I can track
> measures, but mostly in two axes only ( my device is boat autopilot
> computer, should not be flipped :-) )
> Accelerometer is also tricky, however I do have second source ( second
> redundant accelerometer, much more stable and precise )

If you are just using an MPU9250 and host-MCU-based fusion or simple tilt
compensation then you will have to create the dynamic calibration
algorithms yourself. You might take a look at the EM7180 motion
co-processor which has a lot of this dynamic calibration built in.

https://www.tindie.com/products/onehorse/ultimate-sensor-fusion-solution/

jarekkt, 2018-06-05:
Well, I was hoping for easy way - that somebody might have implemented it already.

It seems that I will have to do it myself, I will share results if I get to the point.

jarekkt, 2018-06-05:
Well, I have checked what DMP algorithms do.

It seems that once it detects stable situation on accelerometers it just cancels gyro offsets.

kriswiner, 2018-06-05:
DMP can only do 6 DoF fusion. If you are only worried about gyro offsets,
then detecting no motion and doing gyro offsets on the host is easy to
implement yourself.