kriswiner/MPU9250 #283: sensor config problem

fengrenai, 2018-06-06:
This is my config for MPU9250 via SPI protocol .
  //init mpu9250
  writeSPI(0x6B,0x80);//RESET DEVICE
  delay(1);
  writeSPI(0x6B,0x01);//CLOCK SOURCE
  delay(1);
  writeSPI(0x6C,0x00);//ENABLE ACC AND GYRO 0~2 GYRO 3~5 ACC
  delay(1);
  writeSPI(0x19,0x00);//SAMPLE DATA RATE FOR 1KHZ
  delay(1);
  writeSPI(0x1A,0x01);//BANGWIDTH 184 DELAY2.9MS
  delay(1);
  writeSPI(0x1B,0x18);// GYRO SCALE TO 2000DPS
  delay(1);
  writeSPI(0x1C,0x18);//ACC SCALE TO 2g
  delay(1);
  writeSPI(0X1D,0x01);//DALAY 1.88MS
  delay(1);
  writeSPI(0x37,0x02);
  delay(1);
  writeSPI(0x6A,0x20);//ENABLE FIFO
  delay(1);
  writeSPI(0x23,0x01);//ENABLE SLAVE0
  delay(1);
  writeSPI(0x24,0xDD);//I2C CLOCK 400KHZ
  delay(10);
  writeSPI(0x25,0x18);//SET AK8963 ADD TO I2C//MAYBE 0x18 OR 0x0C
  delay(1);
  writeSPI(0x26,0x0B);//I2C SLAVE 0 REGISTER TO AK8963_CNTL2 
  delay(1);
  writeSPI(0x63,0x01);//RESET AK8963
  delay(1);
  writeSPI(0x27,0x81);//ENABLE I2C AND SET 1 BYTE TO READ
  delay(1);
  writeSPI(0x26,0x0A);//I2C SLAVE 0 REGISTER TO AK8963_CNTL1
  delay(1);
  writeSPI(0x63,0x16);//SET FOR 100HZ
  delay(1);
  writeSPI(0x27,0x81);//ENABLE I2C AND SET 1 BYTE TO READ
  delay(1);
Is there any wrong with my code?

When I try to read the register I have written such like 0x1B, I can only get 0x00 from it. 
Also , I cannot read AK8963 WMI register. And through I can read accl and gyro from register, but it looks like there is some wrong with it.(The three axis gyro data)
[image]

kriswiner, 2018-06-06:
" Is there any wrong with my code? "

I cannot debug your code.

fengrenai, 2018-06-07:
sorry, I haven't put my full code to you.
this is my code for single MPU9250 via SPI  protocol using the Arduino DUE board.
`#include <SPI.h>
#include <I2Cdev.h>

const int CS = 10;

uint8_t   gyro[6],acc[6],aoff[6],goff[6];
int16_t   x_gyro,y_gyro,z_gyro;
int16_t   x_accl,y_accl,z_accl;

byte readSPI(byte addr)
{
  byte result;
  
  select();
  SPI.transfer(addr|0x80);
  result = SPI.transfer(0x00);
  deselect();
  
  return (result);
}

void writeSPI(byte address, byte value)
{
  select();
  SPI.transfer(address);
  SPI.transfer(value);
  deselect();
}

void select(){
  SPI.beginTransaction(SPISettings(1000000,MSBFIRST,SPI_MODE3));
  digitalWrite(CS,LOW);
}

void deselect(){
  digitalWrite(CS,HIGH);
  SPI.endTransaction();
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);

  SPI.begin();
  
  pinMode(CS,OUTPUT);
  digitalWrite(CS, HIGH);

  //init mpu9250
  writeSPI(0x6B,0x80);//RESET DEVICE
  delay(1);
  writeSPI(0x6B,0x01);//CLOCK SOURCE
  delay(1);
  writeSPI(0x6C,0x00);//ENABLE ACC AND GYRO 0~2 GYRO 3~5 ACC
  delay(1);
  writeSPI(0x19,0x00);//SAMPLE DATA RATE FOR 1KHZ
  delay(1);
  writeSPI(0x1A,0x01);//BANGWIDTH 184 DELAY2.9MS
  delay(1);
  writeSPI(0x1B,0x18);// GYRO SCALE TO 2000DPS
  delay(1);
  writeSPI(0x1C,0x18);//ACC SCALE TO 2g
  delay(1);
  writeSPI(0X1D,0x01);//DALAY 1.88MS
  delay(1);
  writeSPI(0x37,0x02);
  delay(1);
  writeSPI(0x6A,0x20);//ENABLE FIFO
  delay(1);
  writeSPI(0x23,0x01);//ENABLE SLAVE0
  delay(1);
  writeSPI(0x24,0xDD);//I2C CLOCK 400KHZ
  delay(10);
  writeSPI(0x25,0x18);//SET AK8963 ADD TO I2C//MAYBE 0x18 OR 0x0C
  delay(1);
  writeSPI(0x26,0x0B);//I2C SLAVE 0 REGISTER TO AK8963_CNTL2 
  delay(1);
  writeSPI(0x63,0x01);//RESET AK8963
  delay(1);
  writeSPI(0x27,0x81);//ENABLE I2C AND SET 1 BYTE TO READ
  delay(1);
  writeSPI(0x26,0x0A);//I2C SLAVE 0 REGISTER TO AK8963_CNTL1
  delay(1);
  writeSPI(0x63,0x16);//SET FOR 100HZ
  delay(1);
  writeSPI(0x27,0x81);//ENABLE I2C AND SET 1 BYTE TO READ
  delay(1);

  //offset
  writeSPI(0x13,0);//gyro
  delay(1);
  writeSPI(0x14,10);
  delay(1);
  writeSPI(0x15,0);
  delay(1);
  writeSPI(0x16,0);
  delay(1);
  writeSPI(0x17,0);
  delay(1);
  writeSPI(0x18,0);
  delay(1);
  writeSPI(0x77,0);//accl
  delay(1);
  writeSPI(0x78,0);
  delay(1);
  writeSPI(0x7A,0);
  delay(1);
  writeSPI(0x7B,0);
  delay(1);
  writeSPI(0x7D,0);
  delay(1);
  writeSPI(0x7E,0);
  delay(1);

  //for mag 
//  uint8_t temp1;
//  writeSPI(0x25,0x18|0x80);
//  delay(10);
//  writeSPI(0x26,0x00);
//  delay(10);
//  writeSPI(0x27,0x81);
//  delay(10);
}

void loop() {
  for(int i = 0 ; i < 6 ; i++){
    gyro[i] = readSPI(0x43 + i);
  }
  x_gyro = (int16_t)(gyro[0] << 8 | gyro[1]);
  y_gyro = (int16_t)(gyro[2] << 8 | gyro[3]);
  z_gyro = (int16_t)(gyro[4] << 8 | gyro[5]);

  for(int i = 0 ; i < 6 ; i++){
    acc[i] = readSPI(0x3B + i);
  }
  x_accl = (int16_t)(acc[0] << 8 | acc[1]);
  y_accl = (int16_t)(acc[2] << 8 | acc[3]);
  z_accl = (int16_t)(acc[4] << 8 | acc[5]);

   //output for original data
  Serial.print(x_gyro);
  Serial.print('\t');
  Serial.print(y_gyro);
  Serial.print('\t');
  Serial.print(z_gyro);
  Serial.print('\t');
  Serial.print(x_accl);
  Serial.print('\t');
  Serial.print(y_accl);
  Serial.print('\t');
  Serial.println(z_accl);
}`

kriswiner, 2018-06-07:
What I meant was that I cannot and will not debug your code. I will answer
your question(s) if you have any.

fengrenai, 2018-06-08:
My question is that I have written value to accl and gyro config，but when I try to read them such as 0x1B, it returns 0x00 to me. Do you have any idea with this situation?  

Thank for your patient and looking forward to hearing from you

fengrenai, 2018-06-08:
I can read 0x71 from the WHO_AM_I register of MPU9250 but I cannot get the 0x48 from the WHO_AM_I register of AK8963C, it returns me 0x00 from register 0x49.

kriswiner, 2018-06-08:
Did you set INT_CFG = 0x02, to turn on pass through mode for the I2C? Are
you using I2C? If you are using SPI, you will have to set up the AK8963C as
a slave I2C device.like this repo does:

https://github.com/kriswiner/MPU9250brianc118

fengrenai, 2018-06-08:
Yes I have written INT_CFG register  0x02, `writeSPI(0x37,0x02); delay(1);` but as I said , when I try to read this register , it only returns 0x00 to me.

kriswiner, 2018-06-08:
Try this here:

https://github.com/kriswiner/MPU9250brianc118/blob/master/examples/MPU9250/MPU9250.ino

and if it doesn't work open an issue.

fengrenai, 2018-06-08:
Ok , it seems like failed connection to mag. I will write issue below that.

Thank you all the same.

kriswiner, 2018-06-08:
The sketch at the link seems to be working using SPI and can talk to the
mag. Have you just tried this sketch? It should work...

fengrenai, 2018-06-08:
I tried it, and it returns 
[image]
It seems like this sketch can read 0x0A  from the WHO_AM_I register of AK8963C.
I don`t know what 0x0A means .

kriswiner, 2018-06-08:
What kind of breakout are you using?

If the hardware is OK, and you have properly selected SPI mode (maybe by
pulling the I2C_select pin LOW) then open an issue and Brian can help
figure out what the trouble is. This sketch should work for any
well-designed MPU9250 breakout board.

fengrenai, 2018-06-08:
I am using SparkFun IMU Breakout - MPU-9250. https://www.sparkfun.com/products/13762.

kriswiner, 2018-06-08:
> I am using SparkFun IMU Breakout - MPU-9250. https://www.sparkfun.com/
> products/13762.

Yes, sparkfun doesn't have an example of how to use their board with SPI.

Please ask Brian Chen for help.

fengrenai, 2018-06-08:
OK, thank you very much.

kriswiner, 2018-06-08:
Yeah, sorry I can;t help more...

kriswiner, 2018-06-08:
> Sorry to bother you again. I have changed my MPU9250,
> https://www.amazon.co.jp/gp/product/B0154PM102/ref=oh_aui_detailpage_o06_s00?ie=UTF8&psc=1.This
> one.
> I can read 0x73 and 0xB0 from the WHO_AM_I register of MPU9250 and AK8963C
> using the Brian Chen`s sketch. Do you have any idea with it?
>
> Looking forward to hearing from you

0x73 means MPU9255, but the AK8963C should still be 0x48.

fengrenai, 2018-06-08:
OK , I solved my problem with the new board. I have see the 0x18 as the WHO_AM_I register of AK8963C. But now I changed it to 0x0C and got the correct value. 

Thank you very much , I owe you a glass of beer.

MitalPattani, 2018-06-08:
@fengrenai hey i have a similar problem with the sparkfun 9250. Can you tell me how you solved your problem? better yet if i could see your code please

MitalPattani, 2018-06-08:
I tried to bypass to access mag reading but i am not able to do it, what could be the possible reasons ?? i tried to put INT_PIN_CNG to 0x12 and 0x22, neither have been working! what could be the possible reasons? i am using sparkfun breakout for 9250 and arduino uno!

kriswiner, 2018-06-08:
Are you using I2C or SPI?

kriswiner, 2018-06-08:
> I switched to spi due to the other parts of the circuit

Did you try the sketches here:
https://github.com/kriswiner/MPU9250brianc118  ?

MitalPattani, 2018-06-08:
That was the basic sketch I used did not get mag readings, then I changed int_pin_cng to 0×22 didn't work, neither did 0×12

MitalPattani, 2018-06-08:
I changed the solder on ic a couple of times is there a chance I could have damaged the sensor?

kriswiner, 2018-06-08:
Always a possibility. Are you sure you have the connections right? You are
able to read the MPU9250 accel/gyro data but not the mag. I think
Sparkfun's board is not designed for SPI.

MitalPattani, 2018-06-08:
Yes the connections are correct and I am getting good raw readings for accel and gyro but not mag

kriswiner, 2018-06-08:
Might want to try another breakout board then...

MitalPattani, 2018-06-08:
I agree

fengrenai, 2018-06-09:
@MitalPattani  Actually I use another breakout rather than Sparkfun's board. My code is on the third floor of this issue. Sparkfun's board is not designed for spiritual protocol for accl,gyro and I2C for mag at the same time I think.

MitalPattani, 2018-06-09:
@fengrenai okay thanks :)