kriswiner/MPU9250 #285: Mahony Quaternion Output Order

jpmsparks, 2018-06-08:
Kris,
Thanks for all your work on this! Quick question - what if the quaternion output order for the Madgwick or MahonyQuaternionUpdate. Is q1==w or does q4 ==w?

``
	norm = sqrt(q1 * q1 + q2 * q2 + q3 * q3 + q4 * q4);
	norm = 1.0f / norm;
	q1 = q1 * norm; 
	q2 = q2 * norm;	
	q3 = q3 * norm;	
	q4 = q4 * norm;	
``

kriswiner, 2018-06-08:
Don't remember but you can easily chack since when the device is pointed
North, q = w == 1, x == 0, y == 0, z == 0.

MitalPattani, 2018-06-10:
Hey, I need to access the quaternions as well, I am using the mahony filter. CPP file! I am accessing the using *getQ, *getQ+1, *getQ+2, *getQ+3, just to be clear values are same as the q1, q2, q3, q4 right? (With reference to the formula mentioned in this Issue)

kriswiner, 2018-06-10:
No idea, you''l have to verify for yourself.