kriswiner/MPU9250 #286: Not enough memory error

avsingh999, 2018-06-10:
Arduino: 1.8.5 (Windows Store 1.8.10.0) (Windows 10), Board: "Arduino/Genuino Uno"

WARNING: library SparkFun_MPU-9250-DMP_Arduino_Library-V_1.0.0 claims to run on (samd) architecture(s) and may be incompatible with your current board which runs on (avr) architecture(s).
Archiving built core (caching) in: C:\Users\AVSING~1\AppData\Local\Temp\arduino_cache_250933\core\core_arduino_avr_uno_a94ab6aaf61dfb93b4a8079c694a14c2.a
Sketch uses 20596 bytes (63%) of program storage space. Maximum is 32256 bytes.

Global variables use 3809 bytes (185%) of dynamic memory, leaving -1761 bytes for local variables. Maximum is 2048 bytes.
Not enough memory; see http://www.arduino.cc/en/Guide/Troubleshooting#size for tips on reducing your footprint.
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

kriswiner, 2018-06-10:
Get  a dev board with a faster MCU with more memory, like this
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
one.

liudr, 2018-06-13:
Were you trying to compile the sparkfun sample code or Kris's sample code? It's good to know which one. Most of the memory cost of either code is from the Serial.print commands. You should be able to save a fair bit of memory by putting the strings in FLASH: Serial.print(F("Hello")) instead of Serial.print("Hello"). This only works on AVR processors, not necessarily on ARM. I am using Arduino Uno with Kris's sample code and I commented out most of the print commands and was using 900 bytes of SRAM. I agree with Kris. Get a better dev board if you need to do a lot more than just running the sample code. Lots of floating-point numbers and operations. AVR is not meant for this.