kriswiner/MPU9250 #287: Randomly can't connect after sketch upload

Genoil, 2018-06-11:
We're using the Sparkfun board and Teensy 3. After sketch upload, the MPU sometimes gets stuck in a state where we can't connect to it any longer. Only a power cycle fixes the issue. Since we're using these in a much bigger system, a power cycle is very costly in terms of time.

When the issue is at hand, I2C SDA seems to be pulled low by the MPU. Is there a way of resetting the board without having to power cycle it?

kriswiner, 2018-06-11:
Sounds like this is a combination of long I2C bus wires (> 10 cm) and weak
Sparkfun pullups (10 K). I would recommend you replace the Sparkfun pullups
with 2K2 pullups and your problem will likely go away, if it is not due to
poor connections.

Genoil, 2018-06-11:
Well there’s only about an inch of solid copper traces on a PCB where both Teensy and MPU are soldered down to. But I’ll add some extra resistors to increase the pull and see what it does. Thanks.

kriswiner, 2018-06-11:
Could also be Sparkfun's crappy MPU9250 board design...