kriswiner/MPU9250 #289: How to calibrate sensors with MPU9250BasicAHRS

tomassalinger, 2018-06-15:
Hi Kris, first of all, thanks a lot for doing this detailed examples.
I have the MPU9255 with arduino pro mini connected by serial usb. 
I'm really new to Arduino, but i managed to get the MPU9250BasicAHRS (Because not using Teensy) showing values but now i'm not sure how could i calibrate the sensors. For what i understood, the BasicAHRS file does not calibrate the sensors, right? and the teensy specific ones do a calibration process. So, how could i calibrate them without using the "t3" files?

Thanks!

kriswiner, 2018-06-15:
Mybe use this one, or make your own version with the calibration functions.

https://github.com/kriswiner/ESP8285/tree/master/EM7180

kriswiner, 2018-06-15:
wrong one, maybe this one:

https://github.com/kriswiner/ESP8285/blob/master/MPU9250/MPU9250_MS5637_BasicAHRS2_ESP8266.ino

tomassalinger, 2018-07-05:
Hi again Kris, i managed to get the MPU9250_MS5637_BasicAHRS2_ESP8266.ino code working for my GY-91 (MPU9255), but i'm not sure if the calibration method is different from the one provided in MPU9250BasicAHRS. I've heard you need a routine to calibrate the sensor, including doing an "8" with the chip in order to calibrate it, is that really necessary or the "factory calibration" provided as is in the code is enough?

Sorry for the newbie questions, i'm trying my best!

Thanks!

kriswiner, 2018-07-05:
You only need to claibrate the sensors if you want accurate absolute
orientation. Otherwise not necessary. But without calibration, of course,
you can get any heading, and probably will.

amishaspatel, 2021-01-05:
Hi Kris. I came across your examples for MPU9250. They are very good, thank you for that.
I too am using MPU9255 with Arduino Nano Every. The WHO_AM_I register return 0x73 HEX value. In order to go ahead with the code do I need to just change the if condition from c == 0x71 to c == 0x73? Are all the functions same for MPU9250 and MPU9255?

Thank you!

kriswiner, 2021-01-05:
Should be enough to change 0x71 to 0x73. The MPU9250 an MPU9255 are very
similar, just that the MPU9255 has some extra embedded functions.