kriswiner/MPU9250 #291: Can not connect to MPU9250 with another I2C device connected

itayrosenml, 2018-06-24:
Hi,

I have no problem connecting to MPU9250 when it's the only I2C device connected. But if another device is connected, I'm getting the following error: `Could not connect to MPU9250: 0x0`.

My code right now is:
`
#include "MPU9250.h"
MPU9250 myIMU;
void setup()
{
  Wire.begin();
  Serial.begin(115200);
  
  // check IMU communication
  byte c = myIMU.readByte(MPU9250_ADDRESS, WHO_AM_I_MPU9250);
  if (c == 0x71)
  {
    myIMU.calibrateMPU9250(myIMU.gyroBias, myIMU.accelBias); // Calibrate gyro and accelerometers, load biases in bias registers
    myIMU.initMPU9250(); // Initialize device for active mode read of acclerometer, gyroscope, and
  }
  else
  {
    Serial.print("Could not connect to MPU9250: 0x");
    Serial.println(c, HEX);
    while(1) ; // Loop forever if communication doesn't happen
  }
}
`

Scanning for I2C devices results in the following:
`
Scanning...
I2C device found at address 0x68  !
done
`

I'm using Arduino mini pro (5v version) and I use a power regulator (by polulu) to provide 3.3v to mpu9250.

kriswiner, 2018-06-24:
No idea. The MPU9250 is not 5 V tolerant, so you likely need logic level
translation for the I2C bus. What is the I2C address of the other device?

itayrosenml, 2018-06-24:
Hi, thanks for your reply!
Turns out they have the same address. After reading the datasheet, looks like I can change MPU92750's address by connecting AD0 to 3.3v. I tried doing so but it's still set to 0x68 rather than 0x69. Am I doing it correctly?

kriswiner, 2018-06-24:
This is sparkfun so you might have to change a solder jumper. Or ask them.

itayrosenml, 2018-06-25:
Hi, I managed to change the device address to 0x69. However, it works only when the address is 0x68. I believe there's some change I should make in the code? I added the following line to my code:
`#define MPU9250_ADDRESS 0x69`. The problem that I'm facing is that the readings are always 0.00 (there's no error).

itayrosenml, 2018-06-25:
I should also note that for some reason it takes 5161 milliseconds to initialize the device now
`
  byte c = myIMU.readByte(0x69, WHO_AM_I_MPU9250);
  if (c == 0x71)
  {
    myIMU.calibrateMPU9250(myIMU.gyroBias, myIMU.accelBias); // Calibrate gyro and accelerometers, load biases in bias registers
    myIMU.initMPU9250(); // Initialize device for active mode read of acclerometer, gyroscope, and
  }
`

kriswiner, 2018-06-25:
That's because the program is actually running and you are calibrating the
sensors, which takes some time.

itayrosenml, 2018-06-26:
Hi, thanks for the reply but it only answers part of my question. I still can't get it to work when using 0x69 address. Am I missing a piece of code I should change?

itayrosenml, 2018-06-27:
I'm getting values of 0 when using address 0x69.

kriswiner, 2018-06-27:
OK, link to sketch being used. At what line do you run into trouble?

itayrosenml, 2018-06-28:
````
#include "MPU9250.h"
MPU9250 myIMU;
void setup()
{
  Wire.begin();
  Serial.begin(115200);
  
  // check IMU communication
  byte c = myIMU.readByte(0x69, WHO_AM_I_MPU9250);
  if (c == 0x71)
  {
    myIMU.calibrateMPU9250(myIMU.gyroBias, myIMU.accelBias); // Calibrate gyro and accelerometers, load biases in bias registers
    myIMU.initMPU9250(); // Initialize device for active mode read of acclerometer, gyroscope, and
  }
  else
  {
    Serial.print("Could not connect to MPU9250: 0x");
    Serial.println(c, HEX);
    while(1) ; // Loop forever if communication doesn't happen
  }
}

void loop() {
 Serial.println(myIMU.ax);
}
````

Serial prints 0's (in void loop).

kriswiner, 2018-06-28:
No idea