kriswiner/MPU9250 #292: Problem with MPU-9250 and RTC DS3231

tobiki, 2018-06-24:
Hello,

I have an MPU-9250 and use your sample "MPU9250BasicAHRS" from the Sparkfun library. In order to use the RTC DS3231, I changed the I2C address of the MPU-9250 in 0x69 by ADO -> 3.3V and
`#define MPU9250_ADDRESS 0x69`
in the source code. If the RTC is not connected, it will work without any problems. With connected RTC this does not work properly. I do not get all the values:

`MPU9250 I AM 71 I should be 71
MPU9250 is online...
x-axis self test: acceleration trim within : 0.0% of factory value
y-axis self test: acceleration trim within : -0.0% of factory value
z-axis self test: acceleration trim within : 0.0% of factory value
x-axis self test: gyration trim within : 0.0% of factory value
y-axis self test: gyration trim within : 0.0% of factory value
z-axis self test: gyration trim within : 0.1% of factory value
MPU9250 initialized for active data mode....
AK8963 I AM 48 I should be 48
AK8963 initialized for active data mode....
X-Axis sensitivity adjustment value 1.18
Y-Axis sensitivity adjustment value 1.18
Z-Axis sensitivity adjustment value 1.14
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -487 my = -95 mz = 519 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00
Yaw, Pitch, Roll: -8.50, 0.00, 0.00
rate = 0.57 Hz
ax = 0.00 ay = 0.00 az = 0.00 mg
gx = 0.00 gy = 0.00 gz = 0.00 deg/s
mx = -494 my = -88 mz = 505 mG
q0 = 1.00 qx = 0.00 qy = 0.00 qz = 0.00`

What is wrong or what can I do? Thanks.

kriswiner, 2018-06-24:
Is the RTC an I2C device? What is its address?

tobiki, 2018-06-24:
Yes. The I2C address is 0x68. That's why I set the MPU address to 0x69.