kriswiner/MPU9250 #293: MPU9250 not detected

OmerKhan13, 2018-06-30:
Hi! 
I can't get my arduino to detect my MPU9250. I added 2 4.7k resistors on the SDA and SCL whilst to VCC(3.4 v) . I've tried almost everything but I cant get it to be detected. the I2C scanner can't detect it on the addresses. Occasionally, I get the first line of all the codes and a seemingly endless "Scanning...." 
I used the basicAHRS code but didn't get anything on the serial monitor. I tried using 5V on it too because It had a voltage regulator on it(not for sda and scl). I tried running without the pull up resistors and vcc and without vcc aswell but nothing seems to work. Any help would be appreciated!

kriswiner, 2018-06-30:
Which MPU9250 breakout and which MCU?

OmerKhan13, 2018-07-01:
Hi Kris,
It's a MPU-92.65(that's what it says on the sensor)
[image]

kriswiner, 2018-07-01:
Did you pull nCS up to enable I2C?

OmerKhan13, 2018-07-03:
Hi Kris, 
I didn't do It I guess. Excuse my question but do I  connect it to 3.3V?

kriswiner, 2018-07-03:
Yes.