kriswiner/MPU9250 #294: Connecting multiple MPU9255s but when I move one I get the reading changes for all of them.

ct14872, 2018-07-08:
Hello Mr. Kris,

First of all let me thank you for your work regarding IMUs and how to get readings from them. 

Right now I have an arduino sketch that takes the readings from a single sensor and performs either the mahony or the madgwick filter and prints the pitch, roll, yaw values on the screen as can be seen in image below 
[image]

Now my problems is that I want to connect more sensors. I have researched a great while before finding your post about the magnetometer address being the same even if the MPUs address changes when pulling AD0 HIGH or LOW. In that post you mentioned that a good way to connect more than a single IMU and get the magnetometer readings from the correct IMU is probably to use an I2C expander like the TCA9548A.

That I did and it worked like a charm. Nonetheless my problem is now arising from a separate problem. I have successfully connected up to 5 MPUs and get different raw readings for each individual MPU. Using your library (basically Sparkfun's but its all taken from your code) I get the problem again by getting all MPUs to affect the same quaternion instead of an individual one. The example serial window below shows that the RAW readings are independent of one another but the quaternion corresponding to each IMU is changing for all IMUs regardles of which one is moving or not:

[image]

This example is from a test using 2 MPU9255 and can be distinguished from the id label on the very left of each serial output line. The raw readings are perfectly individual but the quaternions are exactly the same. 

And here is an example with 2 MPU that can be compared to the first image above where only one is moved:
[image]

I think my problem is developing from the getQ function (from quaternionFilters.ino which gets confused as to which quaternion to return. Am I right by thinking this?

Here is my code:

[kris - Copy.zip](https://github.com/kriswiner/MPU9250/files/2173777/kris.-.Copy.zip)
 Once again thank you for everything and any help would be kindly appreciated as this is going to be part of my dissertation for my master's degree.

All the best,
Constantinos.

kriswiner, 2018-07-08:
Well if you have more than one MPU9250 you will have to have separate q[4]
functions, one for each, separate sensor data a[3], g[3], m[3], etc and
also separate yaw, pitch and roll, one for each, no?

ct14872, 2018-07-08:
Well yeah! Since the MPU9250 is a class, I define an array of MPU9250 each having its own independent variable (ax,ay,az,gx,gy,...) but the quaternion q is different. I am very confused as to why it is defined as static in the quaternionFilters.cpp making it hard for me to wrap my head around the problem here. Should i edit the library so that I pass the q[4] myself you think and include the quaternion q[4] in the MPU9250 class?

kriswiner, 2018-07-08:
You will need to have separate q[4]'s for each MPU9250 sensor; how you do
this is up to you. Easiest way is to define q[n, 4] where n is number of
sensors and pass the q[n, 4] to the function..

ct14872, 2018-07-11:
Hello Mr. Kris,

Sorry for not replying earlier but I have been working on this and have finally got it to work. No I have a serious problem with memory. When I declare an array of MPU9250 of size 4 it seems that it works properly and the data is good. Now when I turn it up a little and ask for 6 MPUs I get a low memory warning and the results are gibberish. Since I am using I2C with a TCA expander can you suggest a way to reduce the memory of your library?

Thank you,
Constantinos.

kriswiner, 2018-07-11:
Maybe use a different MCU, like this
<https://www.tindie.com/products/TleraCorp/dragonfly-stm32l47696-development-board/>
one.

ct14872, 2018-07-11:
well the thing is I can ask for a bigger MCU. But since with 5 it works fine, then there must be something I can do to optimize enough memory so that I can use an extra IMU. I already used the F() macro when printing and even translated all my strings to preallocated character arrays.

I just need another 5-8% of dynamic memory and it will work just fine

Do you believe this is feasible?

kriswiner, 2018-07-11:
No idea