kriswiner/MPU9250 #296: Two MPU9250s Quaternion Issue

benbrown2456, 2018-07-12:
Hi Kris,

I am using two MPUs in the I2C arrangement by switching the AD0 bit to obtain two addresses (0x68 and 0x69). i altered the libraary you have written to ignore the #DEFINE MPU9250_ADDRESS and instead use a variable called 'address' as an input so I can read two at the same time. ]

Example altered library function:
_void MPU9250::readGyroData(int16_t * destination, uint8_t address)
{
  uint8_t rawData[6];  // x/y/z gyro register data stored here
  readBytes(address, GYRO_XOUT_H, 6, &rawData[0]);  // Read the six raw data registers sequentially into data array
  destination[0] = ((int16_t)rawData[0] << 8) | rawData[1] ;  // Turn the MSB and LSB into a signed 16-bit value
  destination[1] = ((int16_t)rawData[2] << 8) | rawData[3] ;  
  destination[2] = ((int16_t)rawData[4] << 8) | rawData[5] ; 
}_

This worked well but as alot of people in this issues section have found the magnetometer readings were the same so I solved this using your suggestion of enabling/disabling the INT_PIN_CFG register to bypass the magnetometer on the IMU I do not want so that it reads the one I do want. This seems to have worked and I am getting different values on the magnetometer for each IMU.

Altered function:
_void MPU9250::readMagData(int16_t * destination, uint8_t address)
{
  //Pick the correct magnetometer
  if (address == 0x68)
  {
	writeByte(0x68, INT_PIN_CFG, 0x22);
	writeByte(0x69, INT_PIN_CFG, 0x20);
  }
  else
  {
	writeByte(0x68, INT_PIN_CFG, 0x20);
	writeByte(0x69, INT_PIN_CFG, 0x22);
  }
  // x/y/z gyro register data, ST2 register stored here, must read ST2 at end of
  // data acquisition
  uint8_t rawData[7];
  // Wait for magnetometer data ready bit to be set
  if(readByte(AK8963_ADDRESS, AK8963_ST1) & 0x01)
  {
    // Read the six raw data and ST2 registers sequentially into data array
    readBytes(AK8963_ADDRESS, AK8963_XOUT_L, 7, &rawData[0]);
    uint8_t c = rawData[6]; // End data read by reading ST2 register
    // Check if magnetic sensor overflow set, if not then report data
    if(!(c & 0x08))
    {
      // Turn the MSB and LSB into a signed 16-bit value
      destination[0] = ((int16_t)rawData[1] << 8) | rawData[0];
      // Data stored as little Endian 
      destination[1] = ((int16_t)rawData[3] << 8) | rawData[2];
      destination[2] = ((int16_t)rawData[5] << 8) | rawData[4];
    }
  }
  // Enable both magnetometers at the end just in case
  writeByte(0x68, INT_PIN_CFG, 0x22);
  writeByte(0x69, INT_PIN_CFG, 0x22);
}_

Now however I am having trouble with the quaternions as my values are the same for each IMU. Here is the code I am using (sorry for the length it also uses an ESP8266 to send the data as a JSON - feel free to use this anyone its mostly amalgamations of code available online).

My theory is that the Quaternion Update might use previously sampled data in which case I might be using the data from both magnetometers to work out the quaternions rather than individually for each however this is not immediately obvious from the quaternionfilters.h library. Maybe I could solve this (if it were the problem by using a pointer in the function to set which MPU9250 object I want to assign each part of the data too rather than currently where I say frontMPU9250 = myMPU9250 however I am not sure how I would do this.

Could you shed any light on this issue?

_#include "quaternionFilters.h"
#include "MPU9250.h"
#include <ESP8266WiFi.h> // Enables the ESP8266 to connect to the local network (via WiFi)
#include <PubSubClient.h> // Allows us to connect to, and publish to the MQTT broker

// Set Up WiFi
const char* ssid = "xyz";
const char* wifi_password = "Obviously not going to post that here!";

// MQTT
[... 488 lines cut ...]
  Serial.print( (int)rearIMU.mx ); Serial.print("\t"); Serial.print("\t");
  Serial.print( (int)rearIMU.my ); Serial.print("\t"); Serial.print("\t");
  Serial.print( (int)rearIMU.mz ); Serial.print("\t"); Serial.print("\t");

  // Rear Quaternions 0,x,y,z
  Serial.print(*getQ()); Serial.print("\t");  Serial.print("\t");
  Serial.print(*(getQ() + 1)); Serial.print("\t");  Serial.print("\t");
  Serial.print(*(getQ() + 2)); Serial.print("\t");  Serial.print("\t");
  Serial.print(*(getQ() + 3)); Serial.print("\t");  Serial.print("\t");

  // Rear IMU Temperature
  Serial.print(rearIMU.temperature, 1); Serial.print("\t");   Serial.print("\t");
  
  // Rear Pitch, Roll, Yaw
  if (calculations)
  {
    Serial.print(rearIMU.pitch, 2); Serial.print("\t");   Serial.print("\t");
    Serial.print(rearIMU.roll, 2); Serial.print("\t");  Serial.print("\t");
    Serial.print(rearIMU.yaw, 2); Serial.print("\t");   Serial.print("\t");
  }

  Serial.print(closecount); Serial.print("\t");   Serial.print("\t");

  Serial.println();
}_

benbrown2456, 2018-07-12:
Was just re-reading my code and I realise that when I print the quaternions I am actually printing the eact same thing *GetQ(), *GetQ()+1 etc which is clearly why I am printing the exact same values. Very obvious and annoyed at myself for not realising this earlier. However my question is now, where are the Quaternions stored and how would I go about storing and accessing the separate values?

kriswiner, 2018-07-12:
Yes, you will need to have two (or n, if you have n MPU9250s on the bus)
versions of the quaternion array q[2,4] to hold the quaternions from each
sensor separately, etc. I don;t have an example of this, but you cam modify
the code accordingly.

MitalPattani, 2018-07-13:
Hey Kris I tried using 2 instances (myIMU [2]) for two IMUs, I am using an i2c multiplexer. Now I am not getting full rotation i.e, if I rotate along the pitch axis by 90 degrees I only get 45 degree on serial!

kriswiner, 2018-07-13:
> Thanks Kris,
>
> As far as I can see q[] is declared as a 1x4 array rather than a 2x4
> (please correct me if I am wrong on that). Does the code remember the last
> quaternions from a previous update? I am wondering if I can simply put
> q1,2,3,4 into my own variables for the front or the rear before I declare
> getData() again so it saves them seperately. This would not work if the
> code took more than just the accel, gyro and mag values as inputs. I feel
> like it should take the previous quaternion values as inputs as it has a
> deltat input however I cannot see in the code exactly where it gets these
> values from.
>
> On a separate note, does the rate at which I sample the values have an
> effect on the accuaracy?

Yes, in the simple example there is only one quaternion. If you want to use
multiple MPU9250 and carry multiple (different) quaternions you will have
to modify the sketch to allow multple quaternions, as in q[n, 4] where n is
the number of MPU9250 you want to track, etc. The quaternions are iterated
in the code using the delta_t time variable.

" On a separate note, does the rate at which I sample the values have an
effect on the accuaracy? "

Yes. Best results are achieved using accle/gyro rates of 1 kHz in my
experience.

kriswiner, 2018-07-13:
Don't know what to tell you, are you using two separate quaternion arrays,
one for each MPU9250?

MitalPattani, 2018-07-15:
That worked, thanks alot.

benbrown2456, 2018-07-16:
Having changed the quarternion code I believe it is working however the main issue is that to toggle the enable bypass bit for the magnetometer I am finding I require a minimum of a 55 millisecond delay between the toggling command and reading the magnetometer leading to my full code having a frequency of around 10 hertz. Any less and I read the wrong magnetometer on occasion (could be 100% of the time with no gap and 50% of the time with a gap of 10 milliseconds). Unless anyone has ever experienced this problem and got past it with some clever technique I have concluded that using SPI or the FIFO bus (not sure if that's the right nomenclature as I have never used it) is the only way to proceed.