kriswiner/MPU9250 #297: MPU9250 Pitch Roll 'Crosstalk'

craigcst, 2018-07-13:
When reading the IMU angle outputs the roll and pitch values are not independent of each other, but they are also not the exact same. Whereas the pitch value IS independent. Any ideas why this 'cross talk' between pitch and roll may be occurring? 

Im Using a MEGA2650 with a MPU9250 on SPI. 

Thanks

craigcst, 2018-07-13:
** YAW value is independant

kriswiner, 2018-07-13:
Did you calibrate your MPU9250? How do you know?

Are you using the proper NED order in the Madgwick/Mahony fusion filter?

Are you running the fusion filter at high enough rate?

bampi2k, 2018-12-31:
Hello @kriswiner ,
I seem to be fighting the same problem.  (I figured that I might be giving the parameters to the quaternion update wrong but just cant get them in the right order/sign)
Using:
1. the latest library from this thread.
2. STM32 'BluePill' from for the processing
3. MPU-92/65 breakout board with the MPU9250 on it.

The way the breakout board is mounted, (positive X in pointing to the forward of the device); The sensor is mounted with components facing up.

My deduction of NED for the sensor was:
Acc = +X(North), -Y(East), -Z(Down)
Gyro = +X, -Y, -Z
Mag = +Y, - X, +Z

[image]

And I am getting the following symptoms:
1. There seems to be a coupling between 2axis... roll-> Yaw and Pitch -> Yaw
2.  Value updates are slow and they settle very slowly.

And yes... I have tried a lot of the NED combinations. Nothing seems tp help.

Could you re-direct me to some resources which tell how to determine the correct input order to the IMU to Madgwick?

kriswiner, 2018-12-31:
Orientation is correct.

Could be two things:

1) poor calibration

2) insufficient fusion rate

I suspect both, but the blue pill is slow like an AVR MCU.

If you run with the accel/gyro sample rates at 200 Hz and mag rate at 100
Hz fou will need a fusion rate (dependent only on the MCU processor speed)
of > 1 kHz and preferably >> 1 kHz to get low latency and high accuracy.

What is the fusion rate you are seeing?

Maybe try one of these
<https://www.tindie.com/products/TleraCorp/ladybug-stm32l432-development-board/>
?

bampi2k, 2018-12-31:
Hi @kriswiner ,

Fusion Rate is between 430Hz to 730Hz.

My Function call looks like:
`    MadgwickQuaternionUpdate(myIMU.ax, -myIMU.ay, -myIMU.az, 
                          myIMU.gx*DEG_TO_RAD, -myIMU.gy*DEG_TO_RAD, -myIMU.gz*DEG_TO_RAD,  
                          myIMU.my, -myIMU.mx, myIMU.mz,myIMU.deltat);`

and then a SerialMonitor log of me just rocking the device on table edge only in the Pitch direction. Notice the drift in Yaw after the operation? 
> Yaw, Pitch, Roll: 66.01, 0.18, -178.11
rate = 490.86 Hz
Yaw, Pitch, Roll: 66.00, 0.41, -178.13
rate = 735.19 Hz
Yaw, Pitch, Roll: 65.98, 0.73, -178.16
rate = 490.14 Hz
Yaw, Pitch, Roll: 65.96, 1.01, -178.21
rate = 734.32 Hz
Yaw, Pitch, Roll: 65.94, 1.38, -178.26
rate = 490.20 Hz
Yaw, Pitch, Roll: 65.92, 1.66, -178.29
rate = 664.34 Hz
Yaw, Pitch, Roll: 65.89, 1.96, -178.31
rate = 664.12 Hz
Yaw, Pitch, Roll: 65.86, 2.38, -178.34
rate = 488.40 Hz
Yaw, Pitch, Roll: 65.84, 2.75, -178.36
rate = 733.57 Hz
Yaw, Pitch, Roll: 65.81, 3.28, -178.39
rate = 439.03 Hz
Yaw, Pitch, Roll: 65.78, 3.84, -178.40
rate = 556.92 Hz
Yaw, Pitch, Roll: 65.76, 4.29, -178.43
rate = 733.25 Hz
Yaw, Pitch, Roll: 65.73, 4.86, -178.48
rate = 488.76 Hz
Yaw, Pitch, Roll: 65.71, 5.35, -178.51
rate = 732.17 Hz
Yaw, Pitch, Roll: 65.68, 5.95, -178.53
rate = 488.22 Hz
Yaw, Pitch, Roll: 65.65, 6.45, -178.54
rate = 731.96 Hz
Yaw, Pitch, Roll: 65.62, 7.07, -178.55
rate = 488.94 Hz
Yaw, Pitch, Roll: 65.60, 7.58, -178.56
rate = 732.17 Hz
Yaw, Pitch, Roll: 65.57, 8.21, -178.57
rate = 489.18 Hz
Yaw, Pitch, Roll: 65.55, 8.74, -178.57
rate = 733.57 Hz
Yaw, Pitch, Roll: 65.54, 9.38, -178.58
rate = 489.18 Hz
Yaw, Pitch, Roll: 65.52, 9.91, -178.59
rate = 731.96 Hz
Yaw, Pitch, Roll: 65.52, 10.56, -178.62
rate = 487.98 Hz
Yaw, Pitch, Roll: 65.51, 11.09, -178.65
rate = 732.71 Hz
Yaw, Pitch, Roll: 65.51, 11.73, -178.72
rate = 488.70 Hz
Yaw, Pitch, Roll: 65.50, 12.25, -178.78
rate = 734.65 Hz
Yaw, Pitch, Roll: 65.46, 12.87, -178.85
rate = 359.24 Hz
Yaw, Pitch, Roll: 65.42, 13.38, -178.88
rate = 733.78 Hz
Yaw, Pitch, Roll: 65.36, 13.98, -178.88
rate = 487.63 Hz
Yaw, Pitch, Roll: 65.31, 14.48, -178.87
rate = 730.03 Hz
[... 312 lines cut ...]
Yaw, Pitch, Roll: 83.55, 3.09, -177.75
rate = 489.48 Hz
Yaw, Pitch, Roll: 83.50, 2.77, -177.77
rate = 734.21 Hz
Yaw, Pitch, Roll: 83.42, 2.39, -177.78
rate = 490.80 Hz
Yaw, Pitch, Roll: 83.36, 2.09, -177.79
rate = 732.60 Hz
Yaw, Pitch, Roll: 83.28, 1.75, -177.78
rate = 490.32 Hz
Yaw, Pitch, Roll: 83.21, 1.48, -177.76
rate = 732.92 Hz
Yaw, Pitch, Roll: 83.13, 1.10, -177.76
rate = 488.22 Hz
Yaw, Pitch, Roll: 83.07, 0.80, -177.76
rate = 734.11 Hz
Yaw, Pitch, Roll: 82.98, 0.51, -177.81
rate = 490.80 Hz
Yaw, Pitch, Roll: 82.90, 0.29, -177.85
rate = 735.40 Hz
Yaw, Pitch, Roll: 82.80, 0.01, -177.86
rate = 490.44 Hz
Yaw, Pitch, Roll: 82.68, -0.18, -177.87

Maybe calibration is off??

bampi2k, 2019-01-06:
I am using the interrupt. I have some other code running along with the AHRS code. I shall have a closer look into that part. What should be the ideal fusion rate?

As for the calibration, I have calibrated only the magnetometer bias. The typical 'wave in all directions for 20sec method'... Any other calibration procedure?

kriswiner, 2019-01-06:
'wave in all directions for 20sec method'.

Probably not good enough. You need to correct for accel and gyro offset
bias as well.

best fusion rate is 5x or 10x the sensor sample rate.