kriswiner/MPU9250 #298: MPU9250 reproducable yaw error

jhergeth, 2018-07-22:
Hello,
as I am using the MPU9250 with an ESP8266 I had to adapt code from different sources.
I am using the code from Kris to calculate the quaterions, an enhanced version of the Sparkfun-Library and the wikipedia code to calc the euler angles.
I get data from the chip at 100 Hz (polling) and calculate the MahonyQuaternion every 3 ms.
I also included the mag sensor calibration from Kris. And I use the correct declination for my place.

The resulting euler angles are stable and reproducable. Rotating the device forth and back gives results in the same angles.
So I am quite happy with that! Thanks for all your postings Kris, they helped a lot.

But............
when I compare the absolute values to the real values (I know how much I turned the device), I get a reprducable error. It seems not to be influnced by calibration and looks like this:
[image]

Around 0° it is roughly 10%, but increases towards 180°.

Any suggestions? 
The result is the same when I use the MadgwickQuaternion calc.
In the Mahony/Madgwick routines are some free parameters Ki and Kp. Should I change them? Or GyroMeasError/~Drift?

Thanks for your work!
Joachim

kriswiner, 2018-07-22:
Not sure what the trouble is. Quats every 3 ms means a fusion rate of ~300
Hz? Not very fast. I think you could benefit from increasing the sensor
sample rate to 400 Hz or even 1 kHz. But the fusion rate is really to slow.
Shoud be 10x the sample rate or so.

I usually get 2 kHz fusion rate with the ESP8285, see:

https://github.com/kriswiner/ESP8285/tree/master/MPU9250

The fusion parameters don't matter...

jhergeth, 2018-07-22:
Thanks for the fast answer.
I will try that.