kriswiner/MPU9250 #299: Roll changes slower when vertically uninitialized!

MitalPattani, 2018-07-22:
In your code you initialize all the values to zero in the setup. Now, when I run the code when the sensor is at 90degrees on the roll axis and then rotate it back to 0 ,the sensor only shows a change of 30 degrees. Any idea on what could be the reason? Also the pitch changes from 0 to 90 and then back or 0 to -90, it doesn't show a full 180 even

kriswiner, 2018-07-22:
Not calibrated?

Data sample rate too low?

Fusion rate too low?

All three?

MitalPattani, 2018-07-22:
It works perfectly well when I rotate it when calibrated in the horizontal plane, as in parallel to ground but when I calibrate it to 0  when it's vertical the rate falls! Same exact code!

kriswiner, 2018-07-22:
Proper magnetometer calibration means the board samples alll directions
many times, not just in one plane.