kriswiner/MPU9250 #300: About pitch roll and yaw values

Imantha123, 2018-07-30:
I want to know in your code can get  angles in a absolute position? Because always i need to get three of values from reference point and every time change pitch roll and yaw angle.
thank you

kriswiner, 2018-07-30:
Yes, you can calculate the Euler angles.

kriswiner, 2018-07-30:
> Hi..I'm using the Arduino DUE and the MPU 9250. I have made the following
> connections with pull up using 10k resisters. After that i upload
> MPU9250BasicAHRS.ino file for get the yaw pitch and roll.
>
> VCC---3.3V
> GND---GND
> SCL---SCL21
> SDA---SDA20
>
> but in my serial monitor has monitor ,
> MPU9250 I AM FF I should be 71
> Could not connect to MPU9250: 0xFF
>
> could you please help me to get the pitch roll and yaw values.
>
> Thank you

What devices show on an I2C scan? Looks like you don;t have the I2C bus set
up correctly.

Imantha123, 2018-07-30:
i2c scan is 0x68. i dont know how to put the i2C bus setup. could you please tell me how do i do?

kriswiner, 2018-07-30:
Good, your I2C bus seems to be working. And if you read the WHO_AM_I
register you get 0xFF? Maybe something wrong with your wire API?

Imantha123, 2018-07-30:
Could you please help me. i am new for arduino .How do read the i WHO_AM_I register? and what is the correct wire API.

kriswiner, 2018-07-30:
Maybe a Sparkfun tutorial would help.

https://learn.sparkfun.com/tutorials/mpu-9250-hookup-guide

Imantha123, 2018-07-30:
dear kriswiner,

My reading of the who who am i register is 0XFF.  what is the correct wire API?

kriswiner, 2018-07-30:
Use the sparkfun tutorial to fix your problem.

Imantha123, 2018-08-01:
hi kriswiner,

I already connect my mpu9250 imu for the arduino due. but roll pitch yaw
values not change.these are fixed. i used MPU9250BasicAHRS.ino file for get
the yaw pitch roll.
This is my serial monitor values. could you please help me to correct this
program. thank you

MPU9250 is online...
x-axis self test: acceleration trim within : -100.0% of factory value
y-axis self test: acceleration trim within : -100.0% of factory value
z-axis self test: acceleration trim within : -100.0% of factory value
x-axis self test: gyration trim within : -100.0% of factory value
y-axis self test: gyration trim within : -100.0% of factory value
z-axis self test: gyration trim within : -100.0% of factory value
MPU9250 initialized for active data mode....
AK8963 I AM FF I should be 48
AK8963 initialized for active data mode....
X-Axis sensitivity adjustment value 0.50
Y-Axis sensitivity adjustment value 0.50
Z-Axis sensitivity adjustment value 0.50
ax = 1938.17 ay = 111.33 az = 0.98 mg
gx = -85.85 gy = 13.92 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 0.02 qx = 0.09 qy = -0.81 qz = 0.58
Yaw, Pitch, Roll: -186.50, -7.60, -109.56
rate = 0.14 Hz
ax = 1938.17 ay = 111.33 az = 0.98 mg
gx = -85.85 gy = 13.92 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 0.60 qx = 0.40 qy = -0.59 qz = 0.36
Yaw, Pitch, Roll: -48.17, -86.12, 99.79
rate = 707.09 Hz
ax = 1938.17 ay = 111.33 az = 0.98 mg
gx = -85.85 gy = 13.92 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 0.61 qx = 0.41 qy = -0.58 qz = 0.36
Yaw, Pitch, Roll: -45.30, -85.67, 96.77
rate = 707.12 Hz
ax = 1938.17 ay = 111.33 az = 0.98 mg
gx = -85.85 gy = 13.92 gz = 0.00 deg/s
mx = -470 my = -120 mz = -125 mG
q0 = 0.61 qx = 0.41 qy = -0.58 qz = 0.36
Yaw, Pitch, Roll: -45.31, -85.67, 96.78
rate = 707.64 Hz

kriswiner, 2018-08-01:
> issuecomment-408946755

Your I2C calls are not working.

AK8963 I AM FF I should be 48

wrong...

Imantha123, 2018-08-01:
> issuecomment-408950727

dear kriswiner,

i was change in to 0X48 (#define AK8963_WHO_AM_I  0x48  )but its not
working. how do i adjust the I2C calls?

kriswiner, 2018-08-01:
> issuecomment-409421275

Please see Sparkfun's MPU9250 tutorial...

kriswiner, 2018-08-01:
What MPU9250 board are you using?

Imantha123, 2018-08-01:
> issuecomment-409422170

i used this board.

kriswiner, 2018-08-01:
> issuecomment-409426377

Which board?

Imantha123, 2018-08-01:
> issuecomment-409428079

MPU9250

kriswiner, 2018-08-01:
> auth/AGY1qgFE7eS-
> issuecomment-409429094

One more  time, which breakout board are you using?

Imantha123, 2018-08-05:
> sorry i don't know about breakout board. i used only arduino DUE board and
> mpu9250 imu sensor.

Dear kriswiner ,

I used your MPU9250_MS5637_AHRS_t3 code for run my arduino due. Finally i
got the yaw pitch roll values. but problem is when the yaw changes its not
accurate. for a example if i change yaw its change and after long time goes
to settle the values. could you please give me a solution.

kriswiner, 2018-08-05:
Faster mcu.

jay6621, 2019-05-12:
close this