kriswiner/MPU9250 #302: MPU9250 Engin on

metehanemlik, 2018-08-01:
I am designing and codding a mini-drone which is more likely crazyflie.  However, mag calibration is a big problem when drone engine on. I am eliminating the motor effect to mag sensor by using some integers. But, these values always changing.  

how can solve this problem please help 

than you

kriswiner, 2018-08-01:
We use the EM7180 for our flight controllers using brushed motors on our 35
g UAVs and don't have any problem getting accurate heading, etc.

See this:
https://www.tindie.com/products/TleraCorp/ladybug-flight-controller/

metehanemlik, 2018-08-01:
ı my design motors are too close to main board

metehanemlik, 2018-08-01:
espcopter

kriswiner, 2018-08-01:
Closer than this:
https://github.com/simondlevy/Hackflight/wiki/L.10-Maiden-flight

In any case, with the EM7180 it doesn't matter how close.

We have this
<https://www.tindie.com/products/onehorse/superfly-hackable-esp8266-flight-controller/>
FC board that also uses the EM7180 but has the ESP8266 as host.

These work fine on very small UAV with brushed motors.

metehanemlik, 2018-08-01:
ı guess there is a real-time teaching system to converge to the correct heading estimate. It is a great work. But, ı cannot use this system. I am already working on my drone system which includes (the heart of drone)esp8266 + mpu9250 over one year. My drone stability and communication are okay but ı couldn't solve the motor calibration problem. I am planning to do swarm drone application. 

Can you help me with the algorithm?

kriswiner, 2018-08-01:
> I am using Arduino ide.

Of course we know how to manage this problem, but you will have to pay us
for our help. If interested please send me an e-mail at tleracorp@gmail.com.

metehanemlik, 2018-08-01:
ı sent